CongducPham/LowCostLoRaGw #277: Loosing Packets

arnestis, 2020-04-06:
Hello Pham,
I have installed the OS on RPi3 with your image. I am using the SX1276 chip as a receiver on the gateway. Αt first, it seemed to work fine but then I discovered that I do not accept all packages in ΤΤΝ as you can see in the image below.
[image]
On the node side, an Arduino Uno with Dragino Hat is used for a library the lmic was used. I have tested the code on another gateway and another repository and I didn't miss any packages. 
In addition, after several hours of operation, the gateway stops working and log stack at this point.

> 2020-04-06T18:49:26.616591> got first framing byte
2020-04-06T18:49:26.617072> --> got LoRa data prefix
2020-04-06T18:49:26.617577> raw format from LoRa gateway
2020-04-06T18:49:26.618087> LoRaWAN?

Regards!

CongducPham, 2020-04-07:
Hi, what code is running on the node side? Then, when you tested on another gateway, what kind of gateway was it?

arnestis, 2020-04-07:
Hello, thanks for responding!!! I tested on the same RPi3 with the same sx1276 but I used the repository below and on another RPi3 with RAK2245.
https://github.com/JaapBraam/single_chan_pkt_fwd

On the node side, I used the code below actually is from examples (TTN-ABP) with ibm_lmic as the library.

`#include <lmic.h>
#include </hal/hal.h>
#include <SPI.h>

// LoRaWAN NwkSKey, network session key
// This is the default Semtech key, which is used by the early prototype TTN
// network.
static const PROGMEM u1_t NWKSKEY[16] = { 0x6F, 0xBB, 0xD2, 0xCA, 0x94, 0x26, 0x54, 0xA9, 0x7A, 0xB4, 0x7A, 0x6F, 0xEA, 0x87, 0xBB, 0xA5 };

// LoRaWAN AppSKey, application session key
// This is the default Semtech key, which is used by the early prototype TTN
// network.
static const u1_t PROGMEM APPSKEY[16] = { 0xF6, 0x15, 0x38, 0x92, 0xEF, 0x57, 0x8E, 0x81, 0xE9, 0x69, 0xDD, 0xCF, 0x76, 0xDF, 0xE3, 0x95 };

// LoRaWAN end-device address (DevAddr)
static const u4_t DEVADDR = 0x260113C6 ; // <-- Change this address for every node!

// These callbacks are only used in over-the-air activation, so they are
// left empty here (we cannot leave them out completely unless
// DISABLE_JOIN is set in config.h, otherwise the linker will complain).
void os_getArtEui (u1_t* buf) { }
void os_getDevEui (u1_t* buf) { }
void os_getDevKey (u1_t* buf) { }

static uint8_t mydata[] = "Hello, world!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 10;

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,
    .dio = {2, 6, 7},
};

void onEvent (ev_t ev) {
    Serial.print(os_getTime());
    Serial.print(": ");
    switch(ev) {
        case EV_SCAN_TIMEOUT:
            Serial.println(F("EV_SCAN_TIMEOUT"));
            break;
        case EV_BEACON_FOUND:
            Serial.println(F("EV_BEACON_FOUND"));
            break;
        case EV_BEACON_MISSED:
            Serial.println(F("EV_BEACON_MISSED"));
            break;
        case EV_BEACON_TRACKED:
            Serial.println(F("EV_BEACON_TRACKED"));
            break;
        case EV_JOINING:
            Serial.println(F("EV_JOINING"));
            break;
        case EV_JOINED:
            Serial.println(F("EV_JOINED"));
            break;
        case EV_RFU1:
            Serial.println(F("EV_RFU1"));
            break;
        case EV_JOIN_FAILED:
[... 116 lines cut ...]
    LMIC_setLinkCheckMode(0);

    // TTN uses SF9 for its RX2 window.
    LMIC.dn2Dr = DR_SF7;

    // Set data rate and transmit power for uplink (note: txpow seems to be ignored by the library)
    

    for (uint8_t i = 0; i < 9; i++) {
      if (i != 0) {
        LMIC_disableChannel(i);
      }
    }
    LMIC_setDrTxpow(DR_SF7,14);
    // Start job
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}`

Finally, I would like to mention that I did tests with the code you suggest on the Arduino folder "Arduino_LoRa_LMIC_ABP_BASIC" with your libraries and I have the same results.

Regards!!!

CongducPham, 2020-04-07:
Well, looks fine. So I don't know. It can happen that some packets are not well received by gateway. Here I do not see any obvious reason.

CongducPham, 2020-04-07:
BTW, the payload has a lot of 0, is that on purposes?