CongducPham/LowCostLoRaGw #278: Stm32L072 + Sx1272 Uart rcv of packets

Arcanum7, 2020-04-23:
Hi! I have only notebook? and no RaspPi.
I can rcv messages by usb CDC from Stm32L072 + Sx1272 only.
How to get  start GW *.py scripts in whis case?

CongducPham, 2020-04-24:
Hi, you can use the STM programmed with the Arduino_LoRa_Gateway sketch (https://github.com/CongducPham/LowCostLoRaGw/tree/master/Arduino/Arduino_LoRa_Gateway). Then use it as a radio bridge as described here: http://cpham.perso.univ-pau.fr/LORA/RPIgateway.html, look at section "7. Tools for testing & build a LoRa gateway for on-the-go tests"

hope that helps, regards,

Arcanum7, 2020-04-24:
it help.
BUT:
1) it need python2  with installed pyserial and requests modules - nowhere it write
2) how to log (python ./log_gw.py) without Dropbox files and folders need?
3) it starts with sudo only
`sudo python 38400SerialToStdout.py /dev/ttyS0 | sudo python ./post_processing_gw.py
`

CongducPham, 2020-04-24:
You can change the log folder in log_gw.py if it is necessary

Arcanum7, 2020-04-25:
ok, i'll try
`python 38400SerialToStdout.py /dev/ttyUSB0 | python ./post_processing_gw.py | python ./log_gw.py`
i need to execute only this string and i will have gateway?

Arcanum7, 2020-04-25:
> You can change the log folder in log_gw.py if it is necessary

it work!

CongducPham, 2020-04-25:
Great to know, did you also successfully flashed your STM with the Arduino_LoRa_Gateway sketch?

Arcanum7, 2020-04-25:
> 
> 
> Great to know, did you also successfully flashed your STM with the Arduino_LoRa_Gateway sketch?

Yes, I did it - it rcv messages from my device good by CDC (usb com port).

can you  answer https://github.com/CongducPham/LowCostLoRaGw/issues/278#issuecomment-619317312

CongducPham, 2020-04-25:
Hi, yes, normally you should have a single-channel gateway, but limited as downlink is not supported. Normally all Python scripts for cloud upload should work.

Arcanum7, 2020-04-26:
> 
> 
> Hi, yes, normally you should have a single-channel gateway, but limited as downlink is not supported. Normally all Python scripts for cloud upload should work.

ok, uplink only by python scritps.
i close this issue, thanks!

closed 2020-04-26 by Arcanum7