CongducPham/LowCostLoRaGw #279: Arduino Nucleo and Python Scripts  DOWNLINK

Arcanum7, 2020-04-26:
Hi everybody!
I have Arduino Nucleo + sx1272 shield + Arduino_LoRa_Gateway_v2.ino  = it works properly!
I have connected it with linux ubuntu 16.04 LTS in virtualbox (windows as host system), crosslink the virtual com port to linux and it work.
do this cmd
`python 38400SerialToStdout.py /dev/ttyS0 | python ./post_processing_gw.py | python ./log_gw.py`
and i can see the UPLINK frames from my devices.
thats good!
Can it do the DOWNLINK?

p.s. i can connect this Gateway to any lorawan server?

CongducPham, 2020-04-26:
Hi, good to hear that it works.

Downlink is thoeretically possible to implement but it is a major effort. You can start with what has been implemented at the higher layer by looking at the scripts/start_gw.sh that launches several tasks, including the one that is realizing PULL DATA from network server. Then, from that you need to send downlink message to your Arduino Nucleo. Modify Arduino_LoRa_Gateway_v2.ino accordingly to handle downlink transmission of the downlink message.

Normally, you should be able connect, a least for uplink, to any lorawan server. You can try to install a local ChirpStack LoRaWAN server and test. See https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-TTN.md

regards,

Arcanum7, 2020-04-28:
If i could do my GW with RPi + spi sx1272 it will  be directional GW ?

CongducPham, 2020-04-28:
Yes, see this README: https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md