CongducPham/LowCostLoRaGw #280: There happens to be some compile errors with lora_gateway.cpp

ga405, 2020-05-06:
`You have a Raspberry 4B
Compiling for Raspberry 4B
g++ -DPABOOST -DMAX_DBM=14 -DGETOPT_ISSUE -DRASPBERRY2 -c lora_gateway.cpp -o lora_gateway_pi2.o
lora_gateway.cpp:351:22: error: ‘loraChannelArray’ was not declared in this scope
 uint32_t loraChannel=loraChannelArray[loraChannelIndex];
                      ^~~~~~~~~~~~~~~~
lora_gateway.cpp:351:22: note: suggested alternative: ‘loraChannel’
 uint32_t loraChannel=loraChannelArray[loraChannelIndex];
                      ^~~~~~~~~~~~~~~~
                      loraChannel
lora_gateway.cpp:351:39: error: ‘loraChannelIndex’ was not declared in this scope
 uint32_t loraChannel=loraChannelArray[loraChannelIndex];
                                       ^~~~~~~~~~~~~~~~
lora_gateway.cpp:351:39: note: suggested alternative: ‘loraChannel’
 uint32_t loraChannel=loraChannelArray[loraChannelIndex];
                                       ^~~~~~~~~~~~~~~~
                                       loraChannel
lora_gateway.cpp: In function ‘int main(int, char**)’:
lora_gateway.cpp:1402:23: error: ‘loraChannelIndex’ was not declared in this scope
                       loraChannelIndex=atoi(optarg);
                       ^~~~~~~~~~~~~~~~
lora_gateway.cpp:1402:23: note: suggested alternative: ‘loraChannel’
                       loraChannelIndex=atoi(optarg);
                       ^~~~~~~~~~~~~~~~
                       loraChannel
lora_gateway.cpp:1403:46: error: ‘STARTING_CHANNEL’ was not declared in this scope
                       if (loraChannelIndex < STARTING_CHANNEL || loraChannelIndex > ENDING_CHANNEL)
                                              ^~~~~~~~~~~~~~~~
lora_gateway.cpp:1403:46: note: suggested alternative: ‘REG_HOP_CHANNEL’
                       if (loraChannelIndex < STARTING_CHANNEL || loraChannelIndex > ENDING_CHANNEL)
                                              ^~~~~~~~~~~~~~~~
                                              REG_HOP_CHANNEL
lora_gateway.cpp:1403:85: error: ‘ENDING_CHANNEL’ was not declared in this scope
                       if (loraChannelIndex < STARTING_CHANNEL || loraChannelIndex > ENDING_CHANNEL)
                                                                                     ^~~~~~~~~~~~~~
lora_gateway.cpp:1403:85: note: suggested alternative: ‘REG_HOP_CHANNEL’
                       if (loraChannelIndex < STARTING_CHANNEL || loraChannelIndex > ENDING_CHANNEL)
                                                                                     ^~~~~~~~~~~~~~
                                                                                     REG_HOP_CHANNEL
lora_gateway.cpp:1405:57: error: ‘STARTING_CHANNEL’ was not declared in this scope
                       loraChannelIndex=loraChannelIndex-STARTING_CHANNEL;
                                                         ^~~~~~~~~~~~~~~~
lora_gateway.cpp:1405:57: note: suggested alternative: ‘REG_HOP_CHANNEL’
                       loraChannelIndex=loraChannelIndex-STARTING_CHANNEL;
                                                         ^~~~~~~~~~~~~~~~
                                                         REG_HOP_CHANNEL
lora_gateway.cpp:1406:35: error: ‘loraChannelArray’ was not declared in this scope
                       loraChannel=loraChannelArray[loraChannelIndex];
                                   ^~~~~~~~~~~~~~~~
lora_gateway.cpp:1406:35: note: suggested alternative: ‘loraChannel’
                       loraChannel=loraChannelArray[loraChannelIndex];
                                   ^~~~~~~~~~~~~~~~
                                   loraChannel
`
Here's the error code.
I cant understand why it happens? Because I'm using rpi4?

CongducPham, 2020-05-07:
in radio.makefile, there needs to be the definition of a frequency band, such as:

CFLAGS=-DPABOOST -DMAX_DBM=14 -DBAND868

see https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/radio.makefile

ga405, 2020-05-07:
Thank you very much, then what is the number ’v‘ if I use a 433MHz model? Still 14? I try again with `CFLAGS=-DPABOOST -DMAX_DBM=14 -DBAND433` then g++ give me these error code:
`g++ -DPABOOST -DMAX_DBM=14 -DBAND433 -DRASPBERRY2 -c lora_gateway.cpp -o lora_gateway_pi2.o
lora_gateway.cpp: In function ‘int main(int, char**)’:
lora_gateway.cpp:1381:39: error: ‘optarg’ was not declared in this scope
            case 'a' : loraMode = atoi(optarg);
                                       ^~~~~~
lora_gateway.cpp:1381:39: note: suggested alternative:
In file included from /usr/include/arm-linux-gnueabihf/bits/getopt_posix.h:27,
                 from /usr/include/unistd.h:869,
                 from arduPi_pi2.h:248,
                 from SX1272.h:36,
                 from lora_gateway.cpp:197:
/usr/include/arm-linux-gnueabihf/bits/getopt_core.h:36:14: note:   ‘unistd::optarg’
 extern char *optarg;
              ^~~~~~
make: *** [makefile:21: lora_gateway_pi2.o] Error 1
`

> in radio.makefile, there needs to be the definition of a frequency band, such as:
> 
> CFLAGS=-DPABOOST -DMAX_DBM=14 -DBAND868
> 
> see https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/radio.makefile

ga405, 2020-05-07:
Got it. 'v' is the number of channel right?

ga405, 2020-05-07:
I tried again with CFLAGS=-DPABOOST -DMAX_DBM=00 -DBAND433 but g++ show me the same error code...

CongducPham, 2020-05-07:
As you have a RPI4, be sure to have uncommented CFLAGS+=-DGETOPT_ISSUE in raspberry.makefile. see https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/raspberry.makefile

ga405, 2020-05-07:
It works! thank you!

CongducPham, 2020-05-07:
Good to hear.
regards