CongducPham/LowCostLoRaGw #283: web admin login page

weiwei19890407, 2020-06-09:
Hi,

When I try to go to admin page, it is required to login. May I know what's the defalt username and password. I have tried the default wifi or pi passwrod, not not work.

Thanks.

CongducPham, 2020-06-10:
- login: admin
- password: loragateway

schroepfer, 2020-08-31:
please can you tell me, how i can reset the web admin password?

CongducPham, 2020-08-31:
See page 25 of https://github.com/CongducPham/tutorials/blob/master/Low-cost-LoRa-GW-web-admin.pdf

regards,

schroepfer, 2020-08-31:
Thanks, but i forgot my **web**-password for log in!

CongducPham, 2020-09-03:
There is an auth file somewhere, let me try to recall where it is.

CongducPham, 2020-09-03:
OK, found it, /etc/gw_web_admin/database.json

schroepfer, 2020-09-03:
Ok, thank you very much!

sbb777, 2020-11-17:
I got compile error like following. any suggestion?

make: *** [makefile:288: SX127X_lora_gateway.o] Error 1
SX12XX_lora_gateway.cpp:1461:39: error: 'optarg' was not declared in this scope

schroepfer, 2021-06-14:
(I forgot my password once again... :-(
Please can you tell me, what to do with the /etc/gw_web_admin/database.json to reset the web-password?

I changed it like this, but it doesn't work.                                  
{
  "username": "pi",
  "password": "loragateway"
}

I changed it like this, but it also doesn't work.                               
{
  "username": "",
  "password": ""
}

I deleted the file, nothing!

Pleas can you help me?