CongducPham/LowCostLoRaGw #287: uCam III Support

rony-vargas, 2020-08-03:
First of all I want to thank you for the hard work done, I started this as University Project, now I am doing more as research/investigation/experimentation side of things.

I am looking through the documentation and I need some guidance on the use of uCAM III library for sending data to your gateway.
I see the current documentation but I can't have a clear picture on  how to adapt, if you just give me some ideas it will be greatly appreciated.

Regards

rony-vargas, 2020-08-27:
I have adapted the code provided in the Arduino Section for LoRa uCamII, removing the XBee parts, and I have made it work until the point it tries to do CAD, but then packets are never sent, here is the output of one run:
`Init uCam test.
Init SX1272 LoRa module: send packets without ACK
SX1276 detected, starting
SX1276 LF/HF calibration
...
Setting power ON: state 0
Setting LoRa mode: state 0
Setting Channel: state 0
Set LoRa Power (dBm) to 14
Setting Power: state 0
Getting Preamble Length: state 0
Preamble Length: 8
Setting LoRa addr 4 : state 0
SX1272 configured Init UARTs for uCam board
--->>> Initializing cam 0

Attempt sync 0

Attempt sync 1
Wait Ack
AA
E
D
0
0
0

Camera has Acked...

                   Waiting for SYNC...

                                      Receiving data. Testing to see if it is SYNC...
     AA
matched 0
D
matched 1
0
matched 2
0
matched 3
0
matched 4
0
matched 5 -success

Camera has SYNCED...
                    Sending ACK for sync
Now we can take images!
                       Cams have synched
InImage memory allocation passed
Ready to encode picture data
--->>> Get reference image from uCam 0
--->>> Synching cam 0

Attempt sync 0
Wait Ack
AA
E
D
1
0
0

Camera has Acked...

                   Waiting for SYNC...

                                      Receiving data. Testing to see if it is SYNC...
     AA
matched 0
D
matched 1
0
matched 2
0
matched 3
0
matched 4
0
matched 5 -success

Camera has SYNCED...
                    Sending ACK for sync
Now we can take images!
                       Initial is being sent
Wait Ack
AA
E
1
2
0
0

INITIAL has been acked...
                         Snapshot is being sent
Wait Ack
AA
E
5
3
0
0

SNAPSHOT has been acked...
                          Get picture is being sent
Wait Ack
AA
E
8
4
0
0
Time for get snapshop : 5
Time for get picture : -9280
Encoding picture data, Quality Factor is : 10
MSS for packetization is : 235
QT ok
00AA54A954 00 0A54A952A

Building packet : 230

Sending packet : 230
--> CarrierSense2: do CAD for DIFS=9CAD
--> DIFS duration 0
###1
--> Channel busy. Waiting until free channel
                                            RRRRRRRRRRRRRRRRRRRRR
`
and this keeps repeating forever without stop

The setup I am using: 
Adafruit Feather STM32F405
Adafruit FeatherWing LoRa S127x (Using 868Mhz Band
uCAM III

CongducPham, 2020-08-27:
Hi, some radio module do have issues with CAD when the radio module is not rebooted appropriately. After flashing the device, try to remove the USB cable, then plug again. If the error persists, try setting uint8_t carrier_sense_method=0; to bypass carrier sense.

regards,