CongducPham/LowCostLoRaGw #288: Arduino Gateway

Samuel-ZDM, 2020-08-26:
Hello.

First I want to congratulate you for the great job.

I have some doubts about the arduino gateway.
My first question is whether I can use another LoRa module, RFM95, and use the same library that is being used in the code, the SX1272.
The second question is whether I can receive packages on the arduino gateway from an end-device with an LMIC code. My idea is to send to The Things Network.

Thank you very much in advance.

CongducPham, 2020-08-27:
Hi, yes, we actually use the RFM95 module. For the second question, yes, please read https://github.com/CongducPham/LowCostLoRaGw/tree/master/Arduino#lorawan-example-and-support

regards,

Samuel-ZDM, 2020-08-27:
Thank you.

The Arduino Gateway accepts OTAA and ABP end devices, correct?

CongducPham, 2020-08-27:
Yes, but with some limitations with the single channel version. Read this: https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md

Samuel-ZDM, 2020-08-27:
Thank you very much.

I'm going to do some tests. If you can't, I'll come back to ask a few more questions :). 
But his explanations helped a lot.

Samuel-ZDM, 2020-09-28:
Hello! 
It's me again
If the gateway is like a relay gateway, will the gateway send a downlink message?

CongducPham, 2020-09-29:
It is not really like a relay, but yes the gw can send downlink message. Did you see the related README? https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-downlink.md

Samuel-ZDM, 2020-09-29:
Thanks for your answer.

If I uncomment GW_RELAY, does the gateway just retransmit the message and the downlink doesn't work, or does it continue to work?

CongducPham, 2020-09-29:
GW_RELAY is an experimental setting to use the gateway code to build a simple relay node. There would be no downlink then.

Samuel-ZDM, 2020-10-01:
Ok.
Thank you.
I emailed you about a question I'm facing. Can we talk over there?

closed 2021-05-04 by Samuel-ZDM