CongducPham/LowCostLoRaGw #289: compile error

sbb777, 2020-11-17:
I can see the WEB pages.
But I got some errors. 
Could I have any suggestion?

------- error messages----
**pi@raspberrypi:~/lora_gateway $ python start_gw.py
sudo ./lora_gateway --bw 125 --cr 5 --sf 12 --freq 920.36 --ndl | python post_processing_gw.py | python log_gw.py
sudo: ./lora_gateway: command not found**
post_processing_gw.py found an alert_conf section
Parsing cloud declarations

pi@raspberrypi:~/lora_gateway $ sudo make
g++ -DPABOOST -DMAX_DBM=14 -DBAND868 -DNRESET=6 -DNSS=10 -I SX12XX -I . -DSX127X                                     -c SX12XX_lora_gateway.cpp -o SX127X_lora_gateway.o
SX12XX_lora_gateway.cpp: In function 'int main(int, char**)':
SX12XX_lora_gateway.cpp:1461:39: error: 'optarg' was not declared in this scope
            case 'a' : loraMode = atoi(optarg);
                                       ^~~~~~
SX12XX_lora_gateway.cpp:1461:39: note: suggested alternative:
In file included from /usr/include/arm-linux-gnueabihf/bits/getopt_posix.h:27,
                 from /usr/include/unistd.h:869,
                 from ./arduPi.h:250,
                 from SX12XX/SX127XLT.h:15,
                 from SX12XX_lora_gateway.cpp:214:
/usr/include/arm-linux-gnueabihf/bits/getopt_core.h:36:14: note:   'unistd::opta                                    rg'
 extern char *optarg;
              ^~~~~~
make: *** [makefile:288: SX127X_lora_gateway.o] Error 1
pi@raspberrypi:~/lora_gateway $ ^C
pi@raspberrypi:~/lora_gateway $ SX12XX_lora_gateway.cpp:1461:39: error: 'optarg' was not declared in this scope
-bash: SX12XX_lora_gateway.cpp:1461:39:: command not found
pi@raspberrypi:~/lora_gateway $             case 'a' : loraMode = atoi(optarg);
-bash: syntax error near unexpected token `:'

CongducPham, 2020-11-17:
Hello, I assume that you did not use the SD card image with all the settings already configured. If you compile on Buster version of Raspbian, you need to uncomment in raspberry.makefile the following line:

#CFLAGS+=-DGETOPT_ISSUE

just remove the #, and then recompile.

sbb777, 2020-11-17:
> Hello, I assume that you did not use the SD card image with all the settings already configured. If you compile on Buster version of Raspbian, you need to uncomment in raspberry.makefile the following line:
> 
> #CFLAGS+=-DGETOPT_ISSUE
> 
> just remove the #, and then recompile.

Thanks a lot for response.
Is this link is right for the configured one : http://cpham.perso.univ-pau.fr/LORA/WAZIUP/raspberrypi-buster-WAZIUP-demo.iso.zip
?

CongducPham, 2020-11-17:
yes

CongducPham, 2020-11-17:
then run a full update from the web interface

sbb777, 2020-11-18:
thanks a lot

closed 2020-11-18 by sbb777