CongducPham/LowCostLoRaGw #290: Arduino 1MHz

Samuel-ZDM, 2020-11-19:
Hello!

I'm using an Atmega328 with 1MHz with RFM95, but I can't send a message to the gateway.
My device sends the first message and then no longer sends it.
I am using the ABP example to send to TTN.

Can you help me?

CongducPham, 2020-11-20:
It might come from a timing problem with the 1MHz board. In LMIC there are macros to define how many tick/seconds you are considering. It is defined in oslmic I think.

Samuel-ZDM, 2020-11-21:
Ok.
Do I need to change the values for these macros?
Is there a formula for finding the right values?

CongducPham, 2020-11-22:
Well I don't remember exactly but it LMIC ticks are based on micros() function, which in turn is based on the frequency of your board. See here for some explanation: https://ucexperiment.wordpress.com/2012/03/17/examination-of-the-arduino-micros-function/. So the goal is to scale correctly so that you get about 16us per os tick. Hope that helps

Samuel-ZDM, 2020-11-23:
Thank you very much!
I'll try to fix this!

closed 2021-07-13 by Samuel-ZDM