CongducPham/LowCostLoRaGw #291: LowCostLoRaGw doesn’t work anymore after update to Version 546

schroepfer, 2020-12-03:
Hello,
after the update to:

Git version: 546. Installed version: 546. Date of current distribution is 2020-12-02 16:57:32.735374198 +0100

the Raspberry doesn’t read the sensors anymore.
The pkt logger shows the last logs before the update, but no new one:

Showing last 20 packet receptions. Refreshing every 5s.
2020-12-02T13:40:35.868805> +++ rxlora[865199]. dst=1 type=0x12 src=5 seq=244 len=23 SNR=8 RSSIpkt=-20 BW=125 CR=4/5 SF=12
2020-12-02T13:39:27.553478> +++ rxlora[865199]. dst=1 type=0x12 src=3 seq=61 len=23 SNR=6 RSSIpkt=-58 BW=125 CR=4/5 SF=12

Before the update the system runs very stable. I changed nothing!

Regards
Juergen

CongducPham, 2020-12-03:
Hi, I've just tried and see no issues. Did you perform basic config and then reboot after the update?

schroepfer, 2020-12-03:
Yes, i did so.
After this i did a full update and basic config too, but this doesn't help.
I looked into all the relevant configs, but nothing is changed.

CongducPham, 2020-12-03:
What raspberry model do you have, and what is the Raspbian version? Buster?

schroepfer, 2020-12-03:
It is a Raspberry Zero W. I installed your: raspberrypi-buster-WAZIUP-demo.iso.zip

CongducPham, 2020-12-03:
OK I see, so the SD card image has been working fine on the RPI0, then you did a full update and it is not working anyore right? It is maybe a compilation issue. Do you have ssh access to your RPI? Could you run scripts/compile_lora_gateway.sh please?

schroepfer, 2020-12-03:
Yes,
yes,
and yes, i will run the script....

schroepfer, 2020-12-03:
The script runs with an error at the end.
I reboot, but it doesn:t help.    :-(

_....
lora_gateway.cpp:(.text+0x1dec): undefined reference to `SX1272::setSF(unsigned char)'
lora_gateway.cpp:(.text+0x1f04): undefined reference to `SX1272::setPacketType(unsigned char)'
lora_gateway.cpp:(.text+0x1fe0): undefined reference to `SX1272::sendPacketTimeout(unsigned char, unsigned char*, unsigned short, unsigned short)'
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'lora_gateway' failed
make: *** [lora_gateway] Error 1
~/lora_gateway/scripts
You should reboot your Raspberry
Bye.
pi@raspi-lora:~/lora_gateway/scripts $_

CongducPham, 2020-12-03:
OK, I see, for reason it compile  with the old lora_gateway.cpp file and old SX1272 lib. I don't know why your makefile has not been updated. To check, could you copy/paste the content of the lora_gateway folder, and then the content of the makefile file please?

schroepfer, 2020-12-03:
i don't unterstand exactly, what i should copy from/to!? I'm no expert in Linux.  :-(

copy the content of /home/pi/lora_gateway of my Raspberry, to.....  paste where?

content of makefile??

CongducPham, 2020-12-03:
When logging with ssh to the RPI, go to the lora_gateway folder type "ls -l" and copy/paste (on this discussion thread) the folder list of file so that I can see how the uodate has been going on. Same for the makefile, type "cat makefile", copy paste the content.

schroepfer, 2020-12-03:
pi@raspi-lora:~/lora_gateway $ ls -l
insgesamt 1976
drwxr-xr-x 3 pi pi   4096 Dez  2 16:57 3GDongle
drwxr-xr-x 3 pi pi   4096 Dez  2 16:57 aes-python-lib
-rw-r--r-- 1 pi pi      7 Dez  3 09:40 arch_compiled.txt
-rw-r--r-- 1 pi pi      7 Dez  3 09:31 arch.txt
-rw-r--r-- 1 pi pi  43990 Dez  2 16:57 arduPi.cpp
-rw-r--r-- 1 pi pi  14535 Dez  2 16:57 arduPi.h
-rw-r--r-- 1 pi pi  35352 Dez  3 09:41 arduPi.o
-rw-r--r-- 1 pi pi  43310 Sep 21 13:27 arduPi_pi2.cpp
-rw-r--r-- 1 pi pi  14101 Sep 21 13:27 arduPi_pi2.h
-rwxr-xr-x 1 pi pi  10739 Dez  2 16:57 base64.c
-rwxr-xr-x 1 pi pi   2165 Dez  2 16:57 base64.h
-rw-r--r-- 1 pi pi   6136 Dez  3 09:41 base64.o
-rw-r--r-- 1 pi pi  77976 Dez  2 16:57 bcm2835.h
-rw-r--r-- 1 pi pi  25160 Dez  2 16:57 bcm2835_v160.h
-rw-r--r-- 1 pi pi  19307 Dez  2 16:57 ChangeLog.md
-rw-r--r-- 1 pi pi   7620 Dez  2 16:57 CloudChirpStack.py
-rw-r--r-- 1 pi pi   6174 Dez  2 16:57 CloudFireBaseAES.py
-rw-r--r-- 1 pi pi   5569 Dez  2 16:57 CloudFireBaseLWAES.py
-rw-r--r-- 1 pi pi   5108 Dez  2 16:57 CloudFireBase.py
-rw-r--r-- 1 pi pi  11927 Dez  2 16:57 CloudGpsFile.py
-rw-r--r-- 1 pi pi   7440 Dez  2 16:57 CloudGroveStreams.py
-rw-r--r-- 1 pi pi   5520 Dez  2 16:57 CloudMongoDB.py
-rw-r--r-- 1 pi pi   8336 Dez  2 16:57 CloudMQTT.py
-rw-r--r-- 1 pi pi   6184 Dez  2 16:57 CloudNodeRed.py
-rw-r--r-- 1 pi pi   7502 Dez  2 16:57 CloudNoInternet.py
-rw-r--r-- 1 pi pi   3767 Dez  2 18:59 clouds.json
-rw-r--r-- 1 pi pi   6206 Dez  2 16:57 CloudSMS.py
-rw-r--r-- 1 pi pi   1976 Dez  2 16:57 clouds_parser.py
-rw-r--r-- 1 pi pi   1019 Dez  2 17:00 clouds_parser.pyc
-rw-r--r-- 1 pi pi  11636 Dez  2 16:57 CloudThingSpeak_old.py
-rw-r--r-- 1 pi pi  14366 Dez  2 16:57 CloudThingSpeak.py
-rw-r--r-- 1 pi pi   7727 Dez  2 16:57 CloudTTN.py
-rw-r--r-- 1 pi pi  18436 Dez  2 16:57 CloudWAZIUP_old.py
-rw-r--r-- 1 pi pi  19171 Dez  2 16:57 CloudWAZIUP.py
-rw-r--r-- 1 pi pi   6193 Dez  2 16:57 CloudWAZIUP_SMS.py
-rwxr-xr-x 1 pi pi  15979 Dez  2 16:57 cmd.sh
drwxr-xr-x 2 pi pi   4096 Jul 10 13:31 config_backup
-rw-r--r-- 1 pi pi   6012 Dez  2 16:57 decrypt_AES.py
-rw-r--r-- 1 pi pi   6018 Dez  2 16:57 decrypt_LoRaWAN.py
-rw-r--r-- 1 pi pi  11001 Dez  2 16:57 decrypt_LSC.py
drwxr-xr-x 2 pi pi   4096 Dez  2 16:57 downlink
-rw-r--r-- 1 pi pi   1227 Dez  3 09:40 gateway_conf.json
-rw-r--r-- 1 pi pi     33 Dez  3 09:40 gateway_id.md5
-rw-r--r-- 1 pi pi     17 Dez  3 09:40 gateway_id.txt
drwxr-xr-x 2 pi pi   4096 Dez  2 16:57 gps
drwxr-xr-x 3 pi pi   4096 Dez  2 16:57 gw_web_admin
-rw-r--r-- 1 pi pi    506 Dez  2 16:57 key_AES.py
-rw-r--r-- 1 pi pi    606 Dez  2 16:57 key_AppKey.py
-rw-r--r-- 1 pi pi    654 Dez  2 16:57 key_ChirpStack.py
-rw-r--r-- 1 pi pi    356 Dez  2 16:57 key_FireBase.py
-rw-r--r-- 1 pi pi    916 Dez  2 16:57 key_GpsFile.py
-rw-r--r-- 1 pi pi    309 Dez  2 16:57 key_GroveStreams.py
-rw-r--r-- 1 pi pi    498 Dez  2 16:57 key_LoRaWAN.py
-rw-r--r-- 1 pi pi   2488 Dez  2 16:57 key_LSC.py
-rw-r--r-- 1 pi pi    924 Dez  2 16:57 key_MQTT.py
-rw-r--r-- 1 pi pi    858 Dez  2 16:57 key_NodeRed.py
-rw-r--r-- 1 pi pi    335 Dez  2 16:57 key_NoInternet.py
-rw-r--r-- 1 pi pi    315 Dez  2 16:57 key_SMS.py
-rw-r--r-- 1 pi pi   1139 Dez  2 16:57 key_ThingSpeak.py
-rw-r--r-- 1 pi pi    965 Dez  2 16:57 key_ThingSpeak.py.save
-rw-r--r-- 1 pi pi    597 Dez  2 16:57 key_TTN.py
-rw-r--r-- 1 pi pi   1318 Dez  2 16:57 key_WAZIUP_old.py
-rw-r--r-- 1 pi pi   1332 Dez  2 16:57 key_WAZIUP.py
-rw-r--r-- 1 pi pi    129 Dez  2 16:57 key_WAZIUP_SMS.py
-rw-r--r-- 1 pi pi   4169 Dez  2 16:57 libSMS.py
-rw-r--r-- 1 pi pi   2690 Dez  2 17:00 libSMS.pyc
lrwxrwxrwx 1 pi pi     26 Dez  3 09:40 log -> /home/pi/Dropbox/LoRa-test
-rw-r--r-- 1 pi pi   5402 Dez  2 16:57 log_gw.py
[... 20 lines cut ...]
-rw-r--r-- 1 pi pi   7405 Dez  2 16:57 README-TTN.md
-rw-r--r-- 1 pi pi   7667 Dez  2 16:57 rfcomm-server.py
drwxr-xr-x 5 pi pi   4096 Dez  2 16:57 scripts
drwxr-xr-x 5 pi pi   4096 Dez  2 16:57 sensors_in_raspi
-rw-r--r-- 1 pi pi   4166 Dez  2 16:57 start_gw.py
-rw-r--r-- 1 pi pi   6029 Dez  2 16:57 SX126X_lora_gateway.h
drwxr-xr-x 2 pi pi   4096 Dez  2 16:57 SX1272
-rwxr-xr-x 1 pi pi 210910 Sep 21 13:26 SX1272.cpp
-rw-r--r-- 1 pi pi  41852 Sep 21 13:26 SX1272.h
-rw-r--r-- 1 pi pi  50144 Dez  2 16:57 SX1272_lora_gateway.cpp
-rwxr-xr-x 1 pi pi 197212 Dez  3 09:41 SX127X_lora_gateway
-rw-r--r-- 1 pi pi   5262 Dez  2 16:57 SX127X_lora_gateway.h
-rw-r--r-- 1 pi pi 233256 Dez  3 09:40 SX127X_lora_gateway.o
-rw-r--r-- 1 pi pi  10749 Dez  2 16:57 SX1280_ranging_slave.cpp
-rw-r--r-- 1 pi pi   2149 Dez  2 16:57 SX1280_ranging_slave.h
-rw-r--r-- 1 pi pi   3375 Dez  2 16:57 SX128X_lora_gateway.h
drwxr-xr-x 2 pi pi   4096 Dez  3 09:41 SX12XX
-rw-r--r-- 1 pi pi  54076 Dez  2 16:57 SX12XX_lora_gateway.cpp
-rw-r--r-- 1 pi pi  13709 Dez  2 16:57 SX12XX_simple_lora_gateway.cpp
-rw-r--r-- 1 pi pi   6641 Dez  2 16:57 SX12XX_simple_lora_gateway.h
-rw-r--r-- 1 pi pi   8757 Dez  2 16:57 SX12XX_simple_lora_transmitter.cpp
-rw-r--r-- 1 pi pi   6251 Dez  2 16:57 SX12XX_simple_lora_transmitter.h
drwxr-xr-x 2 pi pi   4096 Dez  2 16:57 test-folder
drwxr-xr-x 3 pi pi   4096 Dez  2 16:57 ucam-images
pi@raspi-lora:~/lora_gateway $

schroepfer, 2020-12-03:
pi@raspi-lora:~/lora_gateway $ cat makefile
include radio.makefile
include raspberry.makefile

################################
# usage options
# 
# - by defining lora_gateway, loragateway_pi2 or lora_gateway_pi4 target according to
#		* to your radio module (SX126X, SX127X or SX128X)
#		* and low-level lib (Libelium 1272/1276 lib or StuartProject's SX12XX lib)
#   * then:
#			- make lora_gateway, or
# 		- make lora_gateway_pi2, or
# 		- make lora_gateway_pi4
#
# - by calling the specific target
#		* make SX1272_lora_gateway or make SX1272_lora_gateway_pi2 or make SX1272_lora_gateway_pi2 (Libelium 1272/1276 lib)
#		* make SX126X_lora_gateway or make SX126X_lora_gateway_pi2 or make SX126X_lora_gateway_pi4 (StuartProject's SX12XX lib)	
#		* make SX127X_lora_gateway or make SX127X_lora_gateway_pi2 or make SX127X_lora_gateway_pi4 (StuartProject's SX12XX lib)	
#		* make SX128X_lora_gateway or make SX128X_lora_gateway_pi2 or make SX128X_lora_gateway_pi4 (StuartProject's SX12XX lib)	
#
# - then to run the gateway
#		* sudo ./lora_gateway
#
# - there is a simple lora gateway program with fixed LoRa parameters to test the StuartProject's SX12XX lib
#		* make SX126X_simple_lora_gateway or make SX126X_simple_lora_gateway_pi2 or make SX126X_simple_lora_gateway_pi4 (StuartProject's SX12XX lib)	
#		* make SX127X_simple_lora_gateway or make SX127X_simple_lora_gateway_pi2 or make SX127X_simple_lora_gateway_pi4 (StuartProject's SX12XX lib)	
#		* make SX128X_simple_lora_gateway or make SX128X_simple_lora_gateway_pi2 or make SX128X_simple_lora_gateway_pi4 (StuartProject's SX12XX lib)
#
# - then to run the simple gateway
#		* sudo ./simple_lora_gateway
#
################################

################################
# targets
################################

# --------------> RPI1 & RPI0
################################

#/////////////////////////////////
#Libelium SX1272/1276 library – only support SX1272/SX1276 radio modules
#lora_gateway: SX1272_lora_gateway
#/////////////////////////////////
#if moving to SX12XX library
#///// uncomment 1 choice ////////
#lora_gateway: SX126X_lora_gateway
lora_gateway: SX127X_lora_gateway
#lora_gateway: SX128X_lora_gateway
#/////////////////////////////////

#SX1272 lora gateway 
####################
SX1272_lora_gateway: SX1272_lora_gateway.o arduPi.o SX1272/SX1272.o base64.o
	g++ -lrt -lpthread SX1272_lora_gateway.o arduPi.o SX1272/SX1272.o base64.o -o SX1272_lora_gateway
	rm -f lora_gateway
	ln -s SX1272_lora_gateway ./lora_gateway

#SX12XX test gateway 
####################
SX126X_simple_lora_gateway: SX126X_simple_lora_gateway.o arduPi.o SX12XX/SX126XLT.o base64.o
	g++ -lrt -lpthread SX126X_simple_lora_gateway.o arduPi.o SX12XX/SX126XLT.o base64.o -o SX126X_simple_lora_gateway
	rm -f simple_lora_gateway
	ln -s SX126X_simple_lora_gateway ./simple_lora_gateway
	
SX127X_simple_lora_gateway: SX127X_simple_lora_gateway.o arduPi.o SX12XX/SX127XLT.o base64.o
	g++ -lrt -lpthread SX127X_simple_lora_gateway.o arduPi.o SX12XX/SX127XLT.o base64.o -o SX127X_simple_lora_gateway
	rm -f simple_lora_gateway
	ln -s SX127X_simple_lora_gateway ./simple_lora_gateway
[... 281 lines cut ...]
arduPi_pi2.o: arduPi.cpp arduPi.h
	g++ $(CFLAGS) -DRASPBERRY2 -c arduPi.cpp -o arduPi_pi2.o	

arduPi_pi4.o: arduPi.cpp arduPi.h
	g++ $(CFLAGS) -DRASPBERRY4 -c arduPi.cpp -o arduPi_pi4.o

SX1272/SX1272.o: SX1272/SX1272.cpp SX1272/SX1272.h
	g++ $(CFLAGS) -I SX1272 -I . -c SX1272/SX1272.cpp -o SX1272/SX1272.o

SX12XX/SX126XLT.o: SX12XX/SX126XLT.cpp SX12XX/SX126XLT.h SX12XX/SX126XLT_Definitions.h
	g++ $(CFLAGS) -I SX12XX -I . -c SX12XX/SX126XLT.cpp -o SX12XX/SX126XLT.o

SX12XX/SX127XLT.o: SX12XX/SX127XLT.cpp SX12XX/SX127XLT.h SX12XX/SX127XLT_Definitions.h
	g++ $(CFLAGS) -I SX12XX -I . -c SX12XX/SX127XLT.cpp -o SX12XX/SX127XLT.o
	
SX12XX/SX128XLT.o: SX12XX/SX128XLT.cpp SX12XX/SX128XLT.h SX12XX/SX128XLT_Definitions.h
	g++ $(CFLAGS) -I SX12XX -I . -c SX12XX/SX128XLT.cpp -o SX12XX/SX128XLT.o

base64.o: base64.c base64.h
	g++ -c base64.c -o base64.o	

clean:
	cd SX1272; rm -rf *.o; cd ../SX12XX; rm -rf *.o; cd ..
	rm -rf *.o lora_gateway simple_lora_gateway simple_lora_transmitter ranging_slave SX1272_lora_gateway SX1272_lora_gateway_pi* SX12?X_lora_gateway SX12?X_lora_gateway_pi* SX12?X_simple_lora_gateway SX12?X_simple_lora_gateway_pi* SX12?X_simple_lora_transmitter SX12?X_simple_lora_transmitter_pi* SX1280_ranging_slave SX1280_ranging_slave_pi*
pi@raspi-lora:~/lora_gateway $

CongducPham, 2020-12-03:
OK, looks fine, the update was performed as expected. But then when you call scripts/compile_lora_gateway.sh I don't know why it is compiling the old version. Could you copy/paste the whole output when calling scripts/compile_lora_gateway.sh please?

schroepfer, 2020-12-03:
pi@raspi-lora:~/lora_gateway/scripts $ ./compile_lora_gateway.sh
~/lora_gateway ~/lora_gateway/scripts
cd SX1272; rm -rf *.o; cd ../SX12XX; rm -rf *.o; cd ..
rm -rf *.o lora_gateway simple_lora_gateway simple_lora_transmitter ranging_slave SX1272_lora_gateway SX1272_lora_gateway_pi* SX12?X_lora_gateway SX12?X_lora_gateway_pi* SX12?X_simple_lora_gateway SX12?X_simple_lora_gateway_pi* SX12?X_simple_lora_transmitter SX12?X_simple_lora_transmitter_pi* SX1280_ranging_slave SX1280_ranging_slave_pi*
You have a Raspberry Zero W
Compiling for Raspberry Zero W (same as Raspberry 1)
g++ -DPABOOST -DMAX_DBM=14 -DBAND868 -I SX12XX -I . -DSX127X -c SX12XX_lora_gateway.cpp -o SX127X_lora_gateway.o
g++ -DPABOOST -DMAX_DBM=14 -DBAND868 -c arduPi.cpp -o arduPi.o	
g++ -DPABOOST -DMAX_DBM=14 -DBAND868 -I SX12XX -I . -c SX12XX/SX127XLT.cpp -o SX12XX/SX127XLT.o
g++ -c base64.c -o base64.o	
g++ -lrt -lpthread SX127X_lora_gateway.o arduPi.o SX12XX/SX127XLT.o base64.o -o SX127X_lora_gateway
rm -f lora_gateway
ln -s SX127X_lora_gateway ./lora_gateway	
g++     lora_gateway.cpp SX127X_lora_gateway   -o lora_gateway
SX127X_lora_gateway:(.rodata+0x34b8): multiple definition of `rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::MemoryPoolAllocator<rapidjson::CrtAllocator> >::GenericValue(rapidjson::Type)::defaultFlags'
/tmp/ccdWpgnw.o:(.rodata._ZZN9rapidjson12GenericValueINS_4UTF8IcEENS_19MemoryPoolAllocatorINS_12CrtAllocatorEEEEC4ENS_4TypeEE12defaultFlags[_ZZN9rapidjson12GenericValueINS_4UTF8IcEENS_19MemoryPoolAllocatorINS_12CrtAllocatorEEEEC4ENS_4TypeEE12defaultFlags]+0x0): first defined here
SX127X_lora_gateway: In function `__data_start':
:(.data+0x58): multiple definition of `optFQ'
/tmp/ccdWpgnw.o:(.data+0x60): first defined here
SX127X_lora_gateway: In function `_start':
:(.text+0x0): multiple definition of `_start'
/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crt1.o:/build/glibc-P1SmLh/glibc-2.19/csu/../ports/sysdeps/arm/start.S:79: first defined here
SX127X_lora_gateway:(.rodata+0x33b4): multiple definition of `rapidjson::GenericStringRef<char>::emptyString'
/tmp/ccdWpgnw.o:(.rodata._ZN9rapidjson16GenericStringRefIcE11emptyStringE[_ZN9rapidjson16GenericStringRefIcE11emptyStringE]+0x0): first defined here
SX127X_lora_gateway:(.rodata+0x33b8): multiple definition of `void rapidjson::GenericReader<rapidjson::UTF8<char>, rapidjson::UTF8<char>, rapidjson::CrtAllocator>::ParseStringToStream<0u, rapidjson::UTF8<char>, rapidjson::UTF8<char>, rapidjson::GenericStringStream<rapidjson::UTF8<char> >, rapidjson::GenericReader<rapidjson::UTF8<char>, rapidjson::UTF8<char>, rapidjson::CrtAllocator>::StackStream<char> >(rapidjson::GenericStringStream<rapidjson::UTF8<char> >&, rapidjson::GenericReader<rapidjson::UTF8<char>, rapidjson::UTF8<char>, rapidjson::CrtAllocator>::StackStream<char>&)::escape'
/tmp/ccdWpgnw.o:(.rodata._ZZN9rapidjson13GenericReaderINS_4UTF8IcEES2_NS_12CrtAllocatorEE19ParseStringToStreamILj0ES2_S2_NS_19GenericStringStreamIS2_EENS4_11StackStreamIcEEEEvRT2_RT3_E6escape[_ZZN9rapidjson13GenericReaderINS_4UTF8IcEES2_NS_12CrtAllocatorEE19ParseStringToStreamILj0ES2_S2_NS_19GenericStringStreamIS2_EENS4_11StackStreamIcEEEEvRT2_RT3_E6escape]+0x0): first defined here
SX127X_lora_gateway: In function `_fini':
:(.fini+0x0): multiple definition of `_fini'
/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crti.o:/build/glibc-P1SmLh/glibc-2.19/csu/../ports/sysdeps/arm/crti.S:95: first defined here
SX127X_lora_gateway:(.bss+0x16): multiple definition of `checkForLateDownlinkJACC2'
/tmp/ccdWpgnw.o:(.bss+0x2): first defined here
SX127X_lora_gateway: In function `__data_start':
:(.data+0x14): multiple definition of `loraChannelArray'
/tmp/ccdWpgnw.o:(.data+0x4): first defined here
SX127X_lora_gateway:(.rodata+0x34c8): multiple definition of `rapidjson::internal::Pow10(int)::e'
/tmp/ccdWpgnw.o:(.rodata._ZZN9rapidjson8internal5Pow10EiE1e[_ZZN9rapidjson8internal5Pow10EiE1e]+0x0): first defined here
SX127X_lora_gateway:(.bss+0x18): multiple definition of `status_counter'
/tmp/ccdWpgnw.o:(.bss+0x8): first defined here
SX127X_lora_gateway: In function `getGwDateTime()':
:(.text+0x92c): multiple definition of `getGwDateTime()'
/tmp/ccdWpgnw.o:lora_gateway.cpp:(.text+0x6f4): first defined here
SX127X_lora_gateway:(.rodata+0x0): multiple definition of `_IO_stdin_used'
/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crt1.o:(.rodata.cst4+0x0): first defined here
SX127X_lora_gateway:(.bss+0x20): multiple definition of `optSF'
/tmp/ccdWpgnw.o:(.bss+0x18): first defined here
SX127X_lora_gateway:(.bss+0x23): multiple definition of `optRAW'
/tmp/ccdWpgnw.o:(.bss+0x1b): first defined here
SX127X_lora_gateway: In function `__data_start':
:(.data+0x0): multiple definition of `__data_start'
/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/crt1.o:(.data+0x0): first defined here
SX127X_lora_gateway:(.bss+0x28): multiple definition of `rcv_time_ms'
/tmp/ccdWpgnw.o:(.bss+0x20): first defined here
SX127X_lora_gateway:(.bss+0x21): multiple definition of `optCR'
/tmp/ccdWpgnw.o:(.bss+0x19): first defined here
SX127X_lora_gateway:(.bss+0x2c): multiple definition of `rcv_time_tmst'
/tmp/ccdWpgnw.o:(.bss+0x24): first defined here
SX127X_lora_gateway: In function `__data_start':
:(.data+0x60): multiple definition of `optSW'
/tmp/ccdWpgnw.o:(.data+0x68): first defined here
SX127X_lora_gateway:(.bss+0x24): multiple definition of `optHEX'
/tmp/ccdWpgnw.o:(.bss+0x1c): first defined here
[... cut ...]
lora_gateway.cpp:(.text+0x1dec): undefined reference to `SX1272::setSF(unsigned char)'
lora_gateway.cpp:(.text+0x1f04): undefined reference to `SX1272::setPacketType(unsigned char)'
lora_gateway.cpp:(.text+0x1fe0): undefined reference to `SX1272::sendPacketTimeout(unsigned char, unsigned char*, unsigned short, unsigned short)'
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'lora_gateway' failed
make: *** [lora_gateway] Error 1
~/lora_gateway/scripts
You should reboot your Raspberry
Bye.
pi@raspi-lora:~/lora_gateway/scripts $

CongducPham, 2020-12-03:
Strange. for the first lines it is ok:

pi@raspi-lora:~/lora_gateway/scripts $ ./compile_lora_gateway.sh
~/lora_gateway ~/lora_gateway/scripts
cd SX1272; rm -rf .o; cd ../SX12XX; rm -rf .o; cd ..
rm -rf .o lora_gateway simple_lora_gateway simple_lora_transmitter ranging_slave SX1272_lora_gateway SX1272_lora_gateway_pi SX12?X_lora_gateway SX12?X_lora_gateway_pi SX12?X_simple_lora_gateway SX12?X_simple_lora_gateway_pi SX12?X_simple_lora_transmitter SX12?X_simple_lora_transmitter_pi* SX1280_ranging_slave SX1280_ranging_slave_pi*
You have a Raspberry Zero W
Compiling for Raspberry Zero W (same as Raspberry 1)
g++ -DPABOOST -DMAX_DBM=14 -DBAND868 -I SX12XX -I . -DSX127X -c SX12XX_lora_gateway.cpp -o SX127X_lora_gateway.o
g++ -DPABOOST -DMAX_DBM=14 -DBAND868 -c arduPi.cpp -o arduPi.o
g++ -DPABOOST -DMAX_DBM=14 -DBAND868 -I SX12XX -I . -c SX12XX/SX127XLT.cpp -o SX12XX/SX127XLT.o
g++ -c base64.c -o base64.o
g++ -lrt -lpthread SX127X_lora_gateway.o arduPi.o SX12XX/SX127XLT.o base64.o -o SX127X_lora_gateway
rm -f lora_gateway
ln -s SX127X_lora_gateway ./lora_gateway

Normally the script should stop here. Then I don't know why it is trying to continue with:
g++ lora_gateway.cpp SX127X_lora_gateway -o lora_gateway
...

I don't even have this line in the makefile.

Can you try to just type "make SX127X_lora_gateway" in lora_gateway folder and give me the output please.

schroepfer, 2020-12-03:
This ist the result:

pi@raspi-lora:~/lora_gateway $ make SX127X_lora_gateway
make: 'SX127X_lora_gateway' is up to date.

CongducPham, 2020-12-03:
OK, so check that you have a link called lora_gateway that should point to SX127X_lora_gateway. Reboot and test packet reception. Before, check that lora paramters between device and gateway are the same.
regards,

schroepfer, 2020-12-03:
do you mean a symbolic link like this:
lrwxrwxrwx 1 pi pi     26 Dez  3 09:40 log -> /home/pi/Dropbox/LoRa-test

then i don't have a link from lora_gateway to SX127X_lora_gateway.

CongducPham, 2020-12-03:
OK, so type "make clean", then "make SX127X_lora_gateway"

schroepfer, 2020-12-03:
i made this, reboot and it works!
 2020-12-03T13:08:46.571176> +++ rxlora[865200]. dst=1 type=0x12 src=4 seq=1 len=23 SNR=7 RSSIpkt=-58 BW=125 CR=4/5 SF=12

Thank you very much for your time! You are the best, my friend!!!

Regards
Juergen

closed 2020-12-03 by schroepfer

CongducPham, 2020-12-03:
Good to hear, but still could you send me by mail the makefile you have on your RPI please so I could check & see why there is a problem with normal compilation procedure. Mail: Congduc.Pham64@gmail.com

regards,

schroepfer, 2020-12-03:
is this the makefile: 
-rwxr-xr-x 1 pi pi 197212 Dez  3 13:07 SX127X_lora_gateway
?

CongducPham, 2020-12-03:
the file is called "makefile"

schroepfer, 2020-12-03:
thanks! I found it and send you.

CongducPham, 2020-12-03:
OK, thank you, looks fine for me. So don't know why calling scripts/compile_lora_gateway.sh does not work.

schroepfer, 2020-12-07:
Hello,
my loragateway has failed again! :

Date/Time: 2020-12-07T06:40:17
    Radio configuration file is for single channel radio
    last low-level status: 2020-12-06T12:56:17 0m-0d-17h-44min from current date
    last rx: 2020-12-06T13:00:05.312698> +++ rxlora[865200]. dst=1 type=0x12 src=4 seq=25 len=23 SNR=8 RSSIpkt=-66 BW=125 CR=4/5 SF=12

Then i updated to your new GIT version 547.  It didn't help.

Do you think, that it help, when i installl the Gateway completely new?

CongducPham, 2020-12-07:
Maybe, although I didn't figure out why there is an error for your case. So yes, please try a new install to see if it works better. If not, then the previous solution (manual make) is still valid. keep in touch. regards,