CongducPham/LowCostLoRaGw #294: LowCostLoRaGw is not posting to MQTT

schroepfer, 2020-12-08:
I set up a new Raspi 3 with you buster-image, updated and configured MQTT.
But whatever i do, the MQTT is offline.
My own mosquitto and node-red is running fine under 192.168.178.4:1883 on an other raspi.

_Enabled [server offline] |   

MQTT Server | 192.168.178.4:1883 |  
project name | temperatur |  
organization name | zuhause |  
sensor name | sensor |  
source list | 3, 4, 5, 6, 7, 8_

Log:
2020-12-08T09:54:26.628879> --> cloud[1]
2020-12-08T09:54:26.639161> uploading with python CloudMQTT.py
2020-12-08T09:54:26.646594> python CloudMQTT.py "DS/23.06/BAT/3.30" "1,18,4,200,19,7,-58" "125,5,12,865200" "2020-12-08T09:54:25.317505+01:00" "0000B827EB961164"
2020-12-08T09:54:30.806850> MQTT: server unavailable, retrying to connect soon...
2020-12-08T09:54:30.822059> MQTT: server unavailable, retrying to connect soon...
2020-12-08T09:54:30.850286> MQTT: server unavailable, retrying to connect soon...
2020-12-08T09:54:30.864337> MQTT: server unavailable, retrying to connect soon...
2020-12-08T09:54:30.878439> MQTT: not publishing
2020-12-08T09:54:30.892639> --> cloud end

In clouds.json MQTT is enabled:
    {
      "name": "MQTT cloud",
      "script": "python CloudMQTT.py",
      "type": "MQTT on test.mosquitto.org",
      "enabled": true
    },

Pleas can you help me, what is wrong?

Regards
Juergen

CongducPham, 2020-12-08:
Check your MQTT server, because in my case it is working. So try first to ping 192.168.178.4 or use command line mosquitto_pub to check that your server is reachable.

schroepfer, 2020-12-08:
Yes, my MQTT server is working.

CongducPham, 2020-12-08:
You mean that from the RPI acting as a gateway you can reach the RPI acting as MQTT server with a ping command?

schroepfer, 2020-12-08:
I have one RPI ans LoraGateway and an othe RPI (192.167.178.4) as MQTT ans Node-Red Server.
The MQTT ist working fine.

Ping from loragateway to MQTT-Server is working too:

pi@raspi3-lora:~/lora_gateway $ ping 192.168.178.4
PING 192.168.178.4 (192.168.178.4) 56(84) bytes of data.
64 bytes from 192.168.178.4: icmp_seq=1 ttl=64 time=0.563 ms
64 bytes from 192.168.178.4: icmp_seq=2 ttl=64 time=0.488 ms
64 bytes from 192.168.178.4: icmp_seq=3 ttl=64 time=0.550 ms

CongducPham, 2020-12-08:
Can you then try the following command on the gateway:

mosquitto_pub -h 192.167.178.4-p 1883 -t temperatur/zuhause/sensor4/DS -m "23.06"

schroepfer, 2020-12-08:
..the command does not respond.

pi@raspi3-lora:~/lora_gateway $ mosquitto_pub -h 192.167.178.4 -p 1883 -t temperatur/zuhause/sensor4/DS -m "23.06"

CongducPham, 2020-12-08:
So it means that your MQTT server is not up or reachable.

schroepfer, 2020-12-08:
o, the IP you wrote was wrong... not 192.167.178.4 : 192.168.178.4 !

I get the pub in my Node-Red: (sent from the LoraGateway)

8.12.2020, 12:16:38node: Temp*
temperatur/zuhause/sensor4/DS : msg.payload : string[5]
"23.06"

CongducPham, 2020-12-08:
ok sorry. So the whole MQTT chain seems to work in standalone mode. Could you try to remove ":1883" from the server name in key_MQTT.py and try running command:

python CloudMQTT.py "DS/23.06/BAT/3.30" "1,18,4,200,19,7,-58" "125,5,12,865200" "2020-12-08T09:54:25.317505+01:00" "0000B827EB961164"

schroepfer, 2020-12-08:
pi@raspi3-lora:~/lora_gateway $ python CloudMQTT.py "DS/23.06/BAT/3.30" "1,18,4,200,19,7,-58" "125,5,12,865200" "2020-12-08T09:54:25.317505+01:00" "0000B827EB961164"
MQTT: server unavailable, retrying to connect soon...
MQTT: server unavailable, retrying to connect soon...
MQTT: server unavailable, retrying to connect soon...
MQTT: server unavailable, retrying to connect soon...
MQTT: not publishing
pi@raspi3-lora:~/lora_gateway $

schroepfer, 2020-12-08:
and i tested again: and it works.
pi@raspi3-lora:~/lora_gateway $ mosquitto_pub -h 192.168.178.4  -t temperatur/zuhause/sensor4/DS -m "23.06"

CongducPham, 2020-12-08:
ok, maybe I see what happen. CloudMQTT.py tests for server availability using http protocol. Maybe your MQTT server does not run a web server so the http test is returning false. Could you try to replace line:

connected = test_network_available()

in the MQTT_uploadData(nomenclatures, data, src, tdata) function of CloudMQTT.py by:

connected = True

then run again:

python CloudMQTT.py "DS/23.06/BAT/3.30" "1,18,4,200,19,7,-58" "125,5,12,865200" "2020-12-08T09:54:25.317505+01:00" "0000B827EB961164"

and check if your node-red is receiving?

schroepfer, 2020-12-08:
IT WORKED! And i get the Data in Node-Red!

 $ python CloudMQTT.py "DS/23.06/BAT/3.30" "1,18,4,200,19,7,-58" "125,5,12,865200" "2020-12-08T09:54:25.317505+01:00" "0000B827EB961164"
MQTT: publishing
CloudMQTT: will issue cmd
mosquitto_pub -h 192.168.178.4 -t temperatur/zuhause/sensor4/DS -m "23.06"
['mosquitto_pub', '-h', '192.168.178.4', '-t', 'temperatur/zuhause/sensor4/DS', '-m', '23.06']
MQTT: publish success
CloudMQTT: will issue cmd
mosquitto_pub -h 192.168.178.4 -t temperatur/zuhause/sensor4/BAT -m "3.30"
['mosquitto_pub', '-h', '192.168.178.4', '-t', 'temperatur/zuhause/sensor4/BAT', '-m', '3.30']
MQTT: publish success

Thank you very much!

I think, many users are using mosquitto and node-red. Do they all have this problem?

CongducPham, 2020-12-08:
This problem will happen when the computer with the MQTT broker does not have an HTTP server. I will change how the server availability is tested and not using http request, maybe a simple ping.

regards,

CongducPham, 2020-12-19:
Hi, I've changed CloudMQTT.py. You can have a look at the test_network_available() function which uses ping now to test for MQTT server availability.

schroepfer, 2020-12-24:
Thank you! I will test it the next days.

schroepfer, 2020-12-30:
Hello,
i tested your new code and it worked!
Thank you!

CongducPham, 2021-01-03:
Good to hear! Happy New Year 2021!
regards,

closed 2021-04-09 by schroepfer