CongducPham/LowCostLoRaGw #295: GW stopped  trying to send data to ThingSpeak and MQTT cloud

sbb777, 2020-12-09:
on the log the RPI GW doesn't try to send data to  ThingSpeak and MQTT cloud.

Checking the log the RPI GW uploads to CloudTTN and CloudChirpStack, 
but  GW doesn't try to send data to  ThingSpeak and MQTT cloud. (I think I config to upload.)
Which main code is working for main jobs including executing periodic tasks?

On ThingSpeak page/the Admin page 
Date/Time: 2020-12-09T15:43:50

last upload time with CloudThingSpeak.py: 2020-12-01T04:32:21.525779 0m-8d-11h-11min from current date
(on the webpage it says once it uploaed by about 8 days ago, I changed as lorawan mode and enabled TTN and something more but I don't know exactly what I changed and why it stops uploading)

Enabled [server online] |    true
(set for Write key)

====== logs ========================================
2020-12-09T15:17:20.912984> +++ rxlora[868100]. lorawan type=0x40 src=0x26011C80 seq=20 len=10 SNR=8 RSSIpkt=-27 BW=125 CR=4/5 SF=12
2020-12-09T15:17:20.913457> --> number of LoRaWAN enabled clouds is 2
2020-12-09T15:17:20.913830> --> LoRaWAN encrypted cloud[0]
2020-12-09T15:17:20.914182> uploading with python CloudTTN.py
2020-12-09T15:17:20.914530> python CloudTTN.py "QIAcASYAFAABKEz31VUM9DrBmXXoHBk=" "256,64,637607040,20,10,8,-27" "125,5,12,868100" "2020-12-09T15:17:20.898386+09:00*51155282" "0000B827EBFB9E2B"
2020-12-09T15:17:21.260535> CloudTTN: gw id: B827EBFFFFFB9E2B
2020-12-09T15:17:21.261076> CloudTTN: Opening UDP socket to router.eu.thethings.network (52.169.76.203) port 1700...
2020-12-09T15:17:21.261450> CloudTTN: Received packet: {"rxpk": [{"stat": 1, "chan": 0, "datr": "SF12BW125", "tmst": 51155282, "codr": "4/5", "lsnr": 8, "freq": 868.1, "data": "QIAcASYAFAABKEz31VUM9DrBmXXoHBk=", "size": 10, "modu": "LORA", "rfch": 0, "time": "2020-12-09T15:17:20.898386Z", "rssi": -27}]}
2020-12-09T15:17:21.269573> --> LoRaWAN encrypted cloud[1]
2020-12-09T15:17:21.270159> uploading with python CloudChirpStack.py
2020-12-09T15:17:21.270549> python CloudChirpStack.py "QIAcASYAFAABKEz31VUM9DrBmXXoHBk=" "256,64,637607040,20,10,8,-27" "125,5,12,868100" "2020-12-09T15:17:20.898386+09:00*51155282" "0000B827EBFB9E2B"
2020-12-09T15:17:21.602756> CloudChirpStack: gw id: B827EBFFFFFB9E2B
2020-12-09T15:17:21.603554> CloudChirpStack: Opening UDP socket to 127.0.0.1 (127.0.0.1) port 1700...
2020-12-09T15:17:21.609356> --> LoRaWAN encrypted cloud end
2020-12-09T15:17:21.609858> --> DATA encrypted: local AES LoRaWAN not activated
2020-12-09T15:17:21.610207> 
2020-12-09T15:17:22.799478> -----------------------------------------------------
2020-12-09T15:17:22.799952> Check for downlink requests 2020-12-09T15:17:21.79845752055353
2020-12-09T15:17:22.800290> NO NEW DOWNLINK ENTRY
2020-12-09T15:17:22.800635> ----delayRX2-----------------------------------------
2020-12-09T15:17:22.800965> Check for downlink requests 2020-12-09T15:17:22.79876253055658
2020-12-09T15:17:22.801295> NO NEW DOWNLINK ENTRY

====before log (ThingSpeak worked)===========
2020-12-01T04:26:28.118798> --- rxlora. dst=1 type=0x10 src=9 seq=16 len=10 SNR=6 RSSIpkt=-37 BW=125 CR=4/5 SF=12
2020-12-01T04:26:28.119286> 2020-12-01T04:26:28.118521
2020-12-01T04:26:28.119632> rcv ctrl pkt info (^p): 1,16,9,16,10,6,-37
2020-12-01T04:26:28.119972> splitted in:  [1, 16, 9, 16, 10, 6, -37]
2020-12-01T04:26:28.120290> (dst=1 type=0x10(DATA) src=9 seq=16 len=10 SNR=6 RSSI=-37)
2020-12-01T04:26:28.120627> rcv ctrl radio info (^r): 125,5,12,865199
2020-12-01T04:26:28.120954> splitted in:  [125, 5, 12, 865199]
2020-12-01T04:26:28.121451> (BW=125 CR=5 SF=12)
2020-12-01T04:26:28.122232> rcv timestamp (^t): 2020-12-01T04:26:28.116
2020-12-01T04:26:28.122630> 
2020-12-01T04:26:28.122990> adding time zone info
2020-12-01T04:26:28.123314> new rcv timestamp (^t): 2020-12-01T04:26:28.116+01:00
2020-12-01T04:26:28.123789> got first framing byte
2020-12-01T04:26:28.124109> --> got LoRa data prefix
2020-12-01T04:26:28.124425> +++ rxlora[865199]. dst=1 type=0x10 src=9 seq=16 len=10 SNR=6 RSSIpkt=-37 BW=125 CR=4/5 SF=12
2020-12-01T04:26:28.125067> valid app key: accept data
2020-12-01T04:26:28.125555> number of enabled clouds is 1
2020-12-01T04:26:28.125972> --> cloud[0]
2020-12-01T04:26:28.126336> uploading with python CloudThingSpeak.py
2020-12-01T04:26:28.126675> python CloudThingSpeak.py "TC/87.97" "1,16,9,16,10,6,-37" "125,5,12,865199" "2020-12-01T04:26:28.116+01:00" "0000B827EBFB9E2B"
2020-12-01T04:26:30.189756> ThingSpeak: uploading (multiple)
2020-12-01T04:26:30.190343> rcv msg to log (\!) on ThingSpeak ( default , default ): 
2020-12-01T04:26:30.190729> ThingSpeak: will issue curl cmd
2020-12-01T04:26:30.191069> curl -s -k -X POST --data field1=87.97 https://api.thingspeak.com/update?key=SGSH52UGPVAUYG3S
2020-12-01T04:26:30.191665> ThingSpeak: returned code from server is 0, do not retry
2020-12-01T04:26:30.199601> --> cloud end

CongducPham, 2020-12-09:
It seems that your device is a lorawan device so data is encrypted. unless you activate local AES decrypt on the gateway you won't be able to upload to to ThingSpeak. For MQTT, it should work but MQTT cloud should be listed as enabled in the  lorawan cloud section of clouds.json.

sbb777, 2020-12-09:
Thanks for answer. About ThingSpeak
Could I know how to activate the local  AES decrypt on this RPI gateway  if it is posssible?

CongducPham, 2020-12-09:
There are all details in https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-aes_lorawan.md

So you have to set local_aes to true in gateway_conf.json and define the decryption keys in key_LoRaWAN.py.

sbb777, 2021-03-25:
Thanks a lot.

closed 2021-03-25 by sbb777