CongducPham/LowCostLoRaGw #301: Can't change SSID in Wifi Client Mode?

RickerStem, 2021-03-14:
Hi there!

First what a great project! We are already using an ethernet gateway for around 2 years now! We want to expand with some more gateways, but we want to connect them over Wifi. The issue is we can only connect to a Wifi access point with a fresh image? So only with a fresh clean install, we are able to connect to a Wifi access point. But if we change the SSID we want to connect to it doesn't work? It just says:

 "Gateway is normally configured as WiFi client. Connected to: "OurOldConfiguredWiFiAccespoint". Use RaspAP to manage WiFi networks"

So it doesn't switch.. Also sometimes it just says .... Connected to: "Nothing Between The Quotes"  ...... 

Also in RaspAP it says wlan down? and it shows an unknown wifi access point (can't do anything with it)

So only it only works the first time changing the settings, after that it just doesn't work anymore :(

We tried all kind of things:

- Tried to connect through the terminal
- Tried other PI's (tested with 5 different PI's model 3 B v1.2)
- Change WiFi SSIDs to only lowercase
- Update everything as noted on your GIT (also different sequences)
- Used other SD cards 
- Use ethernet cable and then try to configure wifi client
- Waited long enough to see a confirmation message in the admin web interface, then reboot

We're a little bit lost, maybe somebody has suggestions? 

Thanks in advance!

Rick

RickerStem, 2021-03-14:
Some pictures

[image]
[image]
[image]
[image]
[image]

CongducPham, 2021-03-15:
Changing wifi can be problematic because the /etc/wpa_supplicant/wpa_supplicant.conf keeps information about old wifi networks. I believe the simplest way is to delete old wifi line in this file. See some infos here: https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README.md#make-your-gateway-a-wifi-client

RickerStem, 2021-03-15:
Hi,

Thank you for your quick response,

Indeed the wpa_supplicant.conf file was filled with different network settings. I tried to keep only 1 network and then rebooted. But that didn't work. Also tried to remove all the network settings and reboot, also didn't work. after clearing all the network settings and configuring the wifi through your admin web interface, it says it's connected in the WiFi tab, but it's not saying online at the top of the page. also, the test internet connection button still says offline. If I run "iwgetid" it says that I'm connected with "myWiFiSSID" and it shows up in my router and I'm able to ping it?. (I'm also not able to browse the web..)

When running your prepare_wifi_client.sh script I get the following errors. Is this normal?
[image]

So still not able to switch to another wifi.

Rick

CongducPham, 2021-03-15:
I will try to test this evening. I havent' touch the wifi client part for a while as we are mainly using ethernet dhcp.

CongducPham, 2021-03-15:
Well, I tested the wifi and it seems to work fine. 
[image]

CongducPham, 2021-03-15:
The web interface shows that the gw is acting as wifi client and it is online through the wifi connection.
[image]

CongducPham, 2021-03-15:
I have only 1 wifi network indicated in /etc/wpa_supplicant/wpa_supplicant.conf

CongducPham, 2021-03-15:
RaspAP can show the information.
[image]
but actually has some issue to scan other wifi networks.

RickerStem, 2021-03-15:
That's really strange, which raspberry version are you using? (does this maybe matter?) I'm using the Pi 3 Model B V1.2. Also are you able to easily switch to another network? And does it still works? 

Thanks for testing!

RickerStem, 2021-03-15:
I also noticed that it connects to my hotspot when it's booting up? and after a few seconds, it disappears (see video) . Also, some raspberry are really slow with booting up and are hanging a while on raspi-config.service?  

https://user-images.githubusercontent.com/58143786/111181360-2ff01900-85ae-11eb-9b90-564bee2c3d3a.mp4

RickerStem, 2021-03-15:
systemd-analyze blame show 6 minutes for networking service... strange..

[image]

CongducPham, 2021-03-15:
Wifi on RPI can be problematic. It has been reported that when wifi signal is lost then the RPI has some difficulty reconnecting. I'm using an RPI3 B so sale configuration than yours.

RickerStem, 2021-03-16:
Indeed problematic.

I'm now running your gateway on the same Rasberry but now on the newest version of Raspberry OS it's now easy to switch from WiFi SSID. Thankyou for your time and the test!

Going to setup some multiple WiFi gateways and add around 30+ Lora sensors!

CongducPham, 2021-03-16:
Great to hear. regards,

closed 2021-11-19 by RickerStem