CongducPham/LowCostLoRaGw #302: Raspberry PI 4 8G support

man4567890, 2021-03-17:
Hi,
The image raspberrypi-buster-WAZIUP-demo.iso does not seem to work on a Raspberry PI4? Could this be confirmed please

CongducPham, 2021-03-17:
We are haven't tested it on the new RPI 8GB, so maybe there is an unexpected issue. It has been tested extensively on RPI4 2GB and 4GB. What are the errors?

regards,

man4567890, 2021-03-17:
This is what I get with a 32Gb SD card on the console, the exact same SD card boot up in a Raspberry PI 3B+ with no problems

[image]

CongducPham, 2021-03-17:
well, I'm not really able to debug this issue, but it does'nt look like a problem related to RPI4 and amount of RAM memory on it. Try to burn again the sd card image on your 32GB card and run it on the RPI4 to extend the available space.

man4567890, 2021-03-17:
ok will do, just strange that this SD card works on a Pi3B+ but gives this error when running on a Pi4

man4567890, 2021-03-17:
Just to let you know, I re-burned the image with balenaEtcher on a 16G SD card, when I boot the image on a Raspberry PI 4, it gives the same error. I then tried to download the image again, but I get this an error. The image I downloaded yesterday could be corrupted?
[image]

[image]

man4567890, 2021-03-17:
I think the link on this page still points to jessie
[image]

CongducPham, 2021-03-17:
Oh I see, the latest version is on http://cpham.perso.univ-pau.fr/LORA/WAZIUP/raspberrypi-buster-WAZIUP-demo.iso.zip

So this is the one you got and you still have the error?

man4567890, 2021-03-18:
Ahh ok that makes sense, I just loaded the image on a RPi 4 with 4G and it works, but the same image on a RPi 4 with 8G, brings up this while it is trying to boot
[image]