CongducPham/LowCostLoRaGw #303: Can you connect sensors locally to the gateway and save readings in MongoDB?

man4567890, 2021-03-19:
I have some remote sensors, but also have a sense hat on the Raspberry Pi (the gateway). Is it possible to save the values from the sense hat to the local MongoDB as node 1 maybe?
PS Excellent project, thanks for the hard work

CongducPham, 2021-03-19:
Yes, it is possible, look at the line 175 of post_status_processing_gw.py which reads value from a DHT22 sensor connected to the raspberry (to know the temperature inside the gateway box for instance). You can adapt it to your need.

https://github.com/CongducPham/LowCostLoRaGw/blob/9f930cdc8393c5bc76e901757c07bcde70f588f2/gw_full_latest/post_status_processing_gw.py#L175

Happy that you enjoy the project, much appreciated.

regards,

man4567890, 2021-03-23:
Thank you, I will close the issue

closed 2021-03-23 by man4567890