CongducPham/LowCostLoRaGw #305: MongoDB on Zulu (GMT+0) Time

man4567890, 2021-03-24:
Hi
Although my time zone is set to GMT +2, the MongoDB saves the values in Zulu time or GMT
```
pi@iot-gateway01:~/lora_gateway/sensors_in_raspi $ date
Wed 24 Mar 09:01:21 SAST 2021
pi@iot-gateway01:~/lora_gateway/sensors_in_raspi $ 

```
[image]

Anyway to fix that please?

CongducPham, 2021-03-24:
You need to change how time is handle in the python script that saves data in Mongo: CloudMongoDB.py, line 66

man4567890, 2021-03-24:
The change that seems to work is in Mongo: CloudMongoDB.py, line 66
#now = datetime.datetime.utcnow()
now = datetime.datetime.now()

[image]

pi@iot-gateway01:~/lora_gateway $ date
Wed 24 Mar 12:08:29 SAST 2021
pi@iot-gateway01:~/lora_gateway $ 

Thanks for the help!