CongducPham/LowCostLoRaGw #306: LowCostLoRaGw doesn't receive anymore after update to GIT Version 556

schroepfer, 2021-04-09:
Hello,
yesterday i updated my Gateway to your latest version 556.
I run full update, basic conf and reboot.
After this, the gateway receives no more:

    Date/Time: 2021-04-09T06:55:19
    Radio configuration file is for single channel radio

    last low-level status: 2021-04-08T11:50:33 0m-0d-19h-4min from current date
    last rx: 2021-04-08T14:12:17.606256> +++ rxlora[865200]. dst=1 type=0x12 src=5 seq=2 len=23 SNR=7 RSSIpkt=-21 BW=125 CR=4/5 SF=12

schroepfer, 2021-04-09:
i answer myself:
it seems, that this is the same problem that i habe in Dec. 2020.
There you told me to try:
"make clean", then "make SX127X_lora_gateway"

This helped too and it works now again!

Do you have an idea, how to fix this for the next update?

Regards
Juergen

CongducPham, 2021-04-09:
Well, normally on boot the lora gateway radio program should be recompiled for the correct RPI model. There could be some inconsistencies during the update process because local config files are preserved, including radio.makefile and raspberry.makefile. However, when there are changes in those files the update procedure will not be able to replace them automatically, so you need to download them manually.

radio.makefile has changed because we are progressively moving to the SX12XX lib. raspberry.makefile has also changed because we are progressively moving to wiringPi for 64-bit OS. So the file allows you to use either the old ArduPi lib (not working on 64-bit OS) or wiringPI (for 32-bit and 64-bit OS). The default configuration is SX127X lib and ArduPi.

I believe the issue you got come from that inconsistencies therefore make lora_gateway may not work properly, while make SX127X_lora_gateway can work.

However, what is your RPI model?

schroepfer, 2021-04-09:
Raspberry Zero W

CongducPham, 2021-04-09:
OK, so this is why it is actually working! Because on RPI2&3 or RPI4 it would have been make SX127X_lora_gateway_pi2 or make SX127X_lora_gateway_pi4. The new makefile automatically determine the RPI model so make lora_gateway should work.

schroepfer, 2021-04-09:
Thank you for this information!

closed 2021-06-14 by schroepfer