CongducPham/LowCostLoRaGw #307: Arduino Gateway

Samuel-ZDM, 2021-05-04:
Hello!
I have some doubts about the example Arduino_LoRa_SX12XX_Simple_gateway.

When the end device sends the message to the Arduino Gateway, is the gateway able to obtain all the parameters of the device, for example, the keys of the end device? I am thinking about how I can send to the LoRaWAN application without the keys.

I have not done any testing with this code yet, but I would like to know if it is possible to send this data to the application.

Thank you very much in advance.

CongducPham, 2021-05-04:
If you device sends LoRaWAN (encrypted) messages, then the gateway simply receives the encrypted message. You need to implement additional mechanism to get the keys at gateway level.

Samuel-ZDM, 2021-05-04:
If I send an encrypted message to the LoRaWAN application it is not necessary then to get the keys of the end-device, correct?  Just receive the message and send it directly without picking up the keys.  

Thank you so much for your answer.

CongducPham, 2021-05-04:
yes, like a regular gateway. Here, however, as you are using an arduino as gateway, you probably need to get internet access, or forward the output to a computer that has internet access.

Samuel-ZDM, 2021-05-04:
Very cool.  
I was wondering if it is possible to take this packet and send it to a gateway that has internet access or does it not work on LoRaWAN?  Or if in that case it is necessary to have the keys to send to another gateway?

Thanks, and sorry for many questions.