CongducPham/LowCostLoRaGw #309: Add get RSSI into CloudNodeRed

zaiplanet, 2021-05-21:
Hi sir,
- i want get output rssi at cloudnodered.py
- at "def main(ldata, pdata, rdata, tdata, gwid):" i added info rssi into NodeRed_generateData(data, key_NodeRed.sensor_name+src_str, nomenclatures, tdata, RSSI) <-- here
- and at "def NodeRed_generateData(...) i changed to as image
[image]

but while run cloudnodered.py error occurred and failed 
can you help me get more output rssi info for nodered?
i will wait for your reply, thanks you sir
Sorry my english so bad, sir.

CongducPham, 2021-05-21:
You can interactively run the script to debug it. For instance,

python CloudNodeRed.py "TC/22.5/HUM/67" "1,16,36,0,9,8,-45" "125,5,12" "2018-08-05T11:08:52" "00000027EB5A71F7"

for each value, there will be a line generated in nodered/nodered.txt

{"source":"waziup/UPPA/Sensor36","measure":"TC","value":"22.5"}
{"source":"waziup/UPPA/Sensor36","measure":"HUM","value":"67"}

hope that helps

zaiplanet, 2021-05-21:
hi sir, thanks you so much for your reply
and now i can extract output rssi and more output success
thanks you very much Mr C.Pham :)

closed 2021-06-15 by zaiplanet