CongducPham/LowCostLoRaGw #310: Parts List for PCB assembly

argupta23, 2021-06-28:
@CongducPham 
Thanks for sharing the excellent work and the PCB files.
I was wondering if there was a part list (onboard components) available for the various PCB's that one could upload with the Gerber files to get a quote on parts + assembly?
If yes, can you please share them or point me to the link that contains them?

Thanks

CongducPham, 2021-06-29:
Hi, glad that you the PCB are useful for you! No we do not have any part list on the PCB. They are mainly design to easily solder the LoRa modules that you need to buy separately. regards,

argupta23, 2021-06-29:
@CongducPham 
Thanks for the quick response. 
Any thoughts on the SMA connector and Antenna specs one should be looking at to make sure that we get it right?

CongducPham, 2021-06-29:
Well, you have to use the SMA connector that matches the antenna connector. If your antenna is SMA-male, then use an SMA-female connector on the PCB. Same thing if your antenna is RP SMA-male, use an RP SMA-female connector on the PCB. In my case, as I have SMA-male antenna, I'm mostly using SMA-female for the PCB. Then depending on the thickness of the PCB you want to order, take the same thickness for the SMA connector. Example for me: https://fr.aliexpress.com/item/32779564868.html?spm=a2g0s.9042311.0.0.27426c37B8EgFP