CongducPham/LowCostLoRaGw #311: Nano RFM65PCB: Nano 5v safe?

haimiko, 2021-07-12:
The nano is 5v on all digital pins which is not 3.3v safe for the RFM95.  How is  your PCB working without damaging the RFM95?

CongducPham, 2021-07-12:
There is a 3.3V pin on the Nano.

haimiko, 2021-07-12:
That's only for the RFM95 VCC.  The other Nano digital pins are 5v only.
You either need to use a Logic converter or convert the nano to run in 3.3v
mode.  I've downloaded your files and am modifying them to truely support
the nano.  This board will eventually fry the RFM95 digital pins.

haimiko, 2021-07-12:
Here you go.  Have a look at that.
[NanoLoRaRFM95_3vCompat.zip](https://github.com/CongducPham/LowCostLoRaGw/files/6804388/NanoLoRaRFM95_3vCompat.zip)

CongducPham, 2021-07-13:
OK, I see what you meant. I didn't pay attention to this. Thanks for the new PCB you sent me. We are not really using the Nano and mainly use the ProMini in its 3.3V version.

regards,

CongducPham, 2021-07-13:
I will put a notice for the nano.

regards,