CongducPham/LowCostLoRaGw #312: Raspberry gateway with 2 sx1278

zaiplanet, 2021-08-12:
Hi sir!
I running this project on rápberry 3B+ gateway with 1 sx1278 and 2 node also use sx1278.
But while 2 node transmit same time to gateway, gw only received 1 packet from 1 node. 
 Can we set one more sx1278 on second spi on raspberry, sir?
Sorry about my english

CongducPham, 2021-08-17:
It is theoretically possible but managing both radio modules is very tricky. In your scenario, you may have packet collision. You may need to implement simple listen-before-mechanism with random back-off procedure to limit packet collision.

regards,

zaiplanet, 2021-08-17:
Ok sir. Thank u sir

zaiplanet, 2021-08-17:
How i can enable LBT at nodes with sx1278, sir?

CongducPham, 2021-08-18:
You have to implement it yourself using LoRa's CAD mechanism. You can have a look at our modified SX12XX lib and look at the CarrierSense function: https://github.com/CongducPham/LowCostLoRaGw/blob/9f930cdc8393c5bc76e901757c07bcde70f588f2/Arduino/libraries/SX12XX/src/SX127XLT.cpp#L6224

zaiplanet, 2021-08-25:
Thank you very much, sir
I'll check it 👍