CongducPham/LowCostLoRaGw #313: SD Card Image Unavailable

atonughosh, 2021-08-18:
Hi,
Thanks for the great project.

I tried to download the SD Card image but the link doesn't seem to work.

Could you please provide the image link? I will be greatful.

CongducPham, 2021-08-18:
It is probably due to Google Chrome's security. Try with another browser or disable security check. regards,

dimtsam, 2021-10-01:
Hi
I trying to download SD Card image, but its unavailable. I m using 2 different browser with the same result. Is it possible to help me with that?
Thanks a lot

CongducPham, 2021-10-01:
When did you try? I've updated the links to use https. Looks like recent browsers like Chrome do not want to use http anymore. Could you retry? It may happen that first try indicates a timeout, just try a second time.
regards,

dimtsam, 2021-10-01:
Im trying all day today with no results.
Im using chrome and Mozilla on different devices.
Regards

CongducPham, 2021-10-02:
OK, I've revert back to http because there are some issues with our web server. You can copy/paste the link in Firefox, should work now.
regards,

dimtsam, 2021-10-02:
Hi
I got the file. I used Firefox with no problem now.
Thanks a lot