CongducPham/LowCostLoRaGw #315: Using HTCC-AB01 from Heltec with Arduino_LoRa_SX12XX_Demo_Sensor sketch

Unitech, 2021-11-21:
Hi,

Thanks for this great project.
I'm looking to program an HTCC-AB01 with the Arduino_LoRa_SX12XX_Demo_Sensor sketch but I do not manage to get the [NSS/NRESET/DIO1](https://github.com/CongducPham/LowCostLoRaGw/blob/master/Arduino/Arduino_LoRa_SX12XX_Demo_Sensor/SX126X_RadioSettings.h#L17)... pins configured right.

From what I see from the HTCC-AB01 [Schematic Diagram](https://resource.heltec.cn/download/CubeCell/HTCC-AB01/HTCC-AB01_SchematicDiagram.pdf), I have DIO1 on pin 25, DIO2 on pin 24 and DIO3 on pin 9, but it's not right as I get :

```
No device responding
```

Would you have any idea on how to configure the rights pin to make the sketch runs properly ?

Thank you

CongducPham, 2021-11-22:
Well, as the SX1262 is embedded on this board, the NSS and BUSY pins are probably already wired on the SoC. From the schemaric, LoRaNSS seems to be 22/39/P4.3, SPI_RESET to be 27/P5.7 and SPI_BUSY to be 26/P4.7. 

There is a file on their github that shows all the pin naming: https://github.com/HelTecAutomation/CubeCell-Arduino/blob/master/variants/CubeCell-Board/pins_arduino.h

There is probably a file in their github that defines the pins for the embedded LoRa module: https://github.com/HelTecAutomation/CubeCell-Arduino/tree/master/cores

CongducPham, 2021-11-22:
I actually found this: https://github.com/HelTecAutomation/CubeCell-Arduino/blob/master/cores/asr650x/board/inc/board-config.h

CongducPham, 2021-11-22:
However, the problem is that those are not standard pin numbering so there will certainly the need to embed some of their code and pin definition to make it compiling and addressing correctly the SoC pins.

https://github.com/HelTecAutomation/CubeCell-Arduino/tree/master/cores/asr650x/cores, and especially the ASR_Arduino.h file

Unitech, 2021-11-22:
Thanks for helping.

I included the PINS definition in SX126X_RadioSettings.h:

```c                                                                                                      
#define MCU_PINS                                      \                                                   
  P0_0 = 0, P0_1, P0_2, P0_3, P0_4, P0_5, P0_6, P0_7, \                                                   
    P1_0 , P1_1, P1_2, P1_3, P1_4, P1_5, P1_6, P1_7,  \                                                   
    P2_0 , P2_1, P2_2, P2_3, P2_4, P2_5, P2_6, P2_7,  \                                                   
    P3_0 , P3_1, P3_2, P3_3, P3_4, P3_5, P3_6, P3_7,  \                                                   
    P4_0 , P4_1, P4_2, P4_3, P4_4, P4_5, P4_6, P4_7,  \                                                   
    P5_0 , P5_1, P5_2, P5_3, P5_4, P5_5, P5_6, P5_7,  \                                                   
    P6_0 , P6_1, P6_2, P6_3, P6_4, P6_5, P6_6, P6_7,  \                                                   
    P7_0 , P7_1, P7_2, P7_3, P7_4, P7_5, P7_6, P7_7                                                       
                                                                                                          
#define NSS P4_3                                  //select pin on LoRa device                             
#define NRESET P5_7                                //reset pin on LoRa device                             
//we do not connect RFBUSY                                                                                
//if you have a NiceRF SX1262 which has almost the same pinout than the RFM95 then                        
//you can use our ProMini LoRa PCB for RFM95 where RFBUSY (marked DIO2 on the RFM95 PCB) can be connected\
\                                                                                                         
 to D5                                                                                                    
#define RFBUSY P4_7                                //busy pin on LoRa device                              
//we do not connect DIO1 as we use polling method                                                         
//if you have a NiceRF SX1262 which has almost the same pinout than the RFM95 then                        
//you can use our ProMini LoRa PCB for RFM95 where DIO1 can be connected to D3                            
//in that case, comment #define USE_POLLING in SX126XLT.cpp to use DIO1 interrrupt pin                    
//#define DIO1 3                                  //DIO1 pin on LoRa device, used for RX and TX done      
#define DIO1 -1                                  //DIO1 pin on LoRa device, used for RX and TX done       
//*******                                                                                                 
#define DIO2 -1                                 //DIO2 pin on LoRa device, normally not used so set to -1
#define DIO3 -1                                 //DIO3 pin on LoRa device, normally not used so set to -1
#define RX_EN -1                                //pin for RX enable, used on some SX126X devices, set to \
\                                                                                                         
-1 if not used                                                                                            
#define TX_EN -1                                //pin for TX enable, used on some SX126X devices, set to \
\                                                                                                         
-1 if not used                                                                                            
#define SW -1                                   //SW pin on some Dorji LoRa devices, used to power antenn\
\                                                                                                         
a switch, set to -1 if not used                                                                           
```                    

And it seemds to work!

```                                                                                                       
====================================== [SUCCESS] Took 11.44 seconds =====================================\
=                                                                                                         
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, noco\
ntrol, printable, send_on_enter, time                                                                     
--- More details at http://bit.ly/pio-monitor-filters                                                     
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---                                                            
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---                                     
Simple LoRa sensor demo                                                                                   
LoRa Device found                                                                                         
                                                                                                          
SX1262,865200000hz,SF12,BW125000,CR4:5,LDRO_On,SyncWord_0x1424,IQNormal,Preamble_8                        
SX1262,PacketMode_LoRa,Explicit,LNAgain_Boosted                                                           
[... cut ...]
CRC,A808                                                                                                  
LoRa pkt size 10                                                                                          
LoRa pkt seq 0                                                                                            
LoRa Sent in 1323                                                                                         
```                              

But if I want to include some of their deep sleep code, I think I will have to include their library

CongducPham, 2021-11-22:
Ouah, great to see it is working. Actually, did you tried even without including the pin definitions? Because maybe the board definition in the Arduino IDE already does that.

For low power, it might be possible to see what lib they are using, and just use the low power function adapted for the Cortex-M0 micro-controller.

Unitech, 2021-11-22:
Correct, by setting the proper board model in platform.ini file I manage to access to P4_3, P5_7, P4_7 pins without needing to include the PINs definition.

I create a repo with a working example + include of libraries/SX12XX and libraries/U8g2 and it works great.

 I will look at the deepsleep

CongducPham, 2021-11-22:
That's great! Good to see that it is working. Thanks for digging into this board as it is quite nice!

regards,

Unitech, 2021-11-23:
I did not managed yet to make the HTCC-AB01 sleep, but will keep you updated if I managed to.
Thanks again for help

closed 2021-11-23 by Unitech

CongducPham, 2021-11-23:
I saw the low power example. I believe that the code must be structured differently to have low power.
https://github.com/HelTecAutomation/CubeCell-Arduino/tree/master/libraries/Basics/examples/LowPower