CongducPham/LowCostLoRaGw #317: Latest SDCARD Image URL or file access permission is broken

asprakash, 2022-02-07:
Sir,
- The beta version RaspiOS 64Bit version of 16GB & 8GB SDCARD Google Drive Image URLs are getting 403 error. Kindly check it. 
- Current version, 8GB SDCARD image, Debian Buster for 32bit, is not working on RPI 4, Model B, 2GB RAM. Getting start4x.elf file is not compatible error at booting window. Also tried on RPI0W. Getting kernel panic error.

TIA

CongducPham, 2022-02-07:
Hi, thanks for the feedback. issue with 16GB and 8GB has been solved. The issue with the 32-bit version on RPI4 is strange. We have several RPI running the image. Let me check that again. Regards,

asprakash, 2022-02-07:
- The google drive links are fine now.
- Hope the 32bit issues to be solved. 

Thanks

schroepfer, 2022-04-29:
Please can you tell me, which image runs on the neu Raspberry Pi Zero 2 W ?

Regards
Juergen

CongducPham, 2022-05-02:
We didn't try so far actually. Sorry but the RPI Zero is not our main target. I may want to try to install from source by looking at these instructions: https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-manual-install.md

regards,