CongducPham/LowCostLoRaGw #318: no rx found but...

dumasph, 2022-03-18:
**No rx found & LoRa module seems to be detected at both ends**
First of all, thank you for making available this impressive work.
_Bad news : our small group (5 eng. students and I) is not far from absolute beginners
Good news : we are highly motivated_
Despite our efforts, we do not yet manage to communicate between Arduino end device and RPI GW.

- What (already) works ? 

Case 1: Arduino to Arduino LoRa communications running Stuart Robinson’s basic examples 3_LoRa_Transmiter & 4_LoRa_Receiver (obtained from your https://cpham.perso.univ-pau.fr/LORA/HUBIQUITOUS/solution-lab/arduino-lora-tutorial/src/sketch.zip)
Case 2: Also https://github.com/CongducPham/LowCostLoRaGw/tree/master/Arduino/Arduino_LoRa_SX12XX_Simple_gateway running on an Arduino Mega with the same LoRa Dragino shield receives (although with some printing issues) the “HelloWorld” message from 3_LoRa_Transmitter

- What does not (yet) work ?

Case 3: reception with the RPI GW (whatever the emitter)
Case 4: emission from [LowCostLoRaGw/Arduino/Arduino_LoRa_SX12XX_Demo_Sensor/](https://github.com/CongducPham/LowCostLoRaGw) even when trying to receive with the Arduino GW.

- Usefull(?) messages :

msg 1: (RPI GW) through the web admin interface : **no rx found**
msg 2: (RPI GW) post_processing_log.
[220311_post-processing-log.pdf](https://github.com/CongducPham/LowCostLoRaGw/files/8303783/220311_post-processing-log.pdf)
msg 3: When trying to emit with sx12XX_demo_sensor, msg on serial monitor
[220317_ArduinoSerialMonitor.pdf](https://github.com/CongducPham/LowCostLoRaGw/files/8303820/220317_ArduinoSerialMonitor.pdf)

- Available HW : 

RPI3B+ with downloaded OS last updated through the web admin interface 25/02/2022:
LoRa modules : all SX1276 (RFM95W), either soldered onto a Waziup RPI hat breakout, an UCA antenna PCB breakout for miniPRO (also tried) or bought on a [Dragino UNO shield](https://www.dragino.com/products/lora/item/102-lora-shield.html)
Arduino: UNO's, minoPRO's, MEGA

Thank you in advance for any help that could help us overcome this blocking point

CongducPham, 2022-03-18:
Hi, thank you for your interest. It looks like from the device output that it has some problem to reset the lora radio module. That may happen when the REST line is not wired correctly. Check which pin you use for the LoRa RST line and change accordingly in SX127X_RadioSettings.h. It is pin 4 by default.

You can also disable the listen before talk by commenting line 502 in the Arduino_LoRa_SX12XX_Demo_Sensor.ino file: https://github.com/CongducPham/LowCostLoRaGw/blob/d5a733f4c5f798d0a1f525ce241d46e4b89be72d/Arduino/Arduino_LoRa_SX12XX_Demo_Sensor/Arduino_LoRa_SX12XX_Demo_Sensor.ino#L502

Hope that helps.

regards,

dumasph, 2022-03-18:
Thks for this rapid reply. We will try (probably next thursday). I am travelling. As far as I remember, at least for the Dragino hat, one student set NRESET to 9 in the code (probably because this was the case in Stuart's).
If we use the miniPRO PCB with integrated antenna, we will have less wiring possibilities/error issues and should leave NRESET to 4. Do I understand right ?

dumasph, 2022-03-24:
Your hint was right. It was indeed a problem of bad wiring the RESET. We are now transmitting in LoRa from several Sensor nodes and receiving the frames on the RPI GW or Arduino GW. Now we can go ahead and tackle the communication into the cloud and finalise the sensor side. Many thanks again. 
regards,

CongducPham, 2022-03-24:
Good to know, good luck.
regards

closed 2022-03-31 by dumasph