CongducPham/LowCostLoRaGw #320: Wakeup every 10sec. (Arduino_LoRa_Simple_GPS.ino)

schroepfer, 2022-05-06:
Hello,
thanks for your great project and your fast help in the past!

I'm using an Atmega328 with an GPS and your latest _Arduino_LoRa_Simple_GPS.ino_ with **no** changes.
It works fine, but it wakes up every 10 seconds after sleep, instead of 20 Minutes. (Successfully switch LoRa module in sleep mode)
The GPS-Fix is great.

Please can you tell me, what can be wrong?

Regards
Juergen

benner, 2022-05-06:
Not sure if this the case but actually Arduino/Atmega does multiple sleeps in 8s or 4s to reach total minutes of sleep:

https://github.com/CongducPham/LowCostLoRaGw/blob/master/Arduino/Arduino_LoRa_Simple_GPS/Arduino_LoRa_Simple_GPS.ino#L831

schroepfer, 2022-05-06:
i know this, but:
in CongDucs program he defines nCycle for this loop, but he never used it, (and there is no loop) like in his other code.
`unsigned int nCycle = IDLE_PERIOD_SECONDS/LOW_POWER_PERIOD;`

CongducPham, 2022-05-06:
the new low power control structure is for more accuracy as a sleep time of 8s is actually closest to 8158ms

I actually don't see why your device is waking up every 10s from the sleep loop. I will try to reproduce the error.

regards,

schroepfer, 2022-05-07:
Excuse, i found my misstake!
My GPS-Switching-Pin is 8, not 7 !
So the GPS never switches off and goes not to sleep.
`#define GPS_PIN_POWER 
#define GPS_PIN_POWER_PIN 7
`

Regards,
Juergen

closed 2022-05-07 by schroepfer