CongducPham/LowCostLoRaGw #323: next step to many end devices

dimtsam, 2023-01-21:
Hi
I just made a LoRa gateway using a Raspberry Pi3 and 2 end devices using Arduino Nano
I want to make about 50 end devices to communnicate with the gateway and to send data to Thingspeak cloud. 
What are the steps I must make so each end device sends to specific channel and field at Thingspeak cloud? Now, the 2 end devices that I made sends data, and both of them updates only 1 field at channel I declare to my gateway. I need to "control" which end device sends data to more than 1 channels and fields.
Must modificate .py files at the structure of the gateway at the SD card?
Hope you understand what I am talking about because my programming knowledge is very low
Thanks in advance

CongducPham, 2023-01-23:
Hi, what you need is documented here: https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-NewCloud.md#indicate-a-list-of-allowed-source-addresses-and-other-advanced-features

what you need to define are key_association, field_association and nomenclature_association.

hope that helps.
regards,

dimtsam, 2023-01-25:
I just read the tutiorials and Its exactly what I was looking for. Can you please advice me what is the best way to modify these files? Are located to gw_full_latest? Does it needs any permissions for the modification?
Thanks a lot

CongducPham, 2023-01-26:
On your RPI gateway, these files are in the lora_gateway folder. You can modify them without any problem.
Hope that helps,