CongducPham/LowCostLoRaGw #324: Forward packets to private Chirpstack server

RickerStem, 2023-02-28:
Hi,

At first, we are running your gateway already for 3 years steadily! Great project, thank you.

We are now expanding to LoRaWAN and a private Chirpstack server. We are running your gateway in mode 1. I read your documentation and you use the pkt forwarder. I tried to set the gw to LoRaWAN mode but then the sensors are not working anymore ofc. Furthermore I added the chirpstack to the cloud.json and changed the IP adress in the key_chirpstack.py to connect to my chirpstack server. In the post processing output (option 5 in cmd.sh) I'm seeing the processing of my custom scripts that are called, but I'm not seeing anything of the chirpstack script? Do you have any suggestions? 

So is it possible to forward the non LoRaWAN compatible packets to chirpstack? Or do I also need LoRaWAN code on my sensors?

*Official Dragino gateway is working in Chirpstack.

Thank you in advance.

Rick

CongducPham, 2023-03-01:
3 years! that's great! thank you for the feedback. Well, it is simpler to have LoRaWAN on your sensor node because the network server expects a LoRaWAN packet (in terms of packet format). Although it could be possible to wrap the non-LoRaWAN packet from sensors in a LoRaWAN packet at the gateway, that would need the knowledge of encryption keys,... It is not really advised to do so.

regards,

RickerStem, 2023-03-03:
Thank you for your quick response, I think I will go for the simpler option :)
I will look into your LoRaWAN examples and change it to that software.

tk5ep, 2023-07-16:
@RickerStem 

I'm almost in the same situation as you.
I've build a gateway, installed a private Chirpstack server. So far, the gateway appears online.

I have partially wired a low power board with a PCB found on this GitHub and I'm waiting for LoRa modules to complete the board and gateway.

I'm in the learning process and understand that I need to use LoRaWan on my sensors.

I found the 2 LoRaWan examples here and others on other sites but I'm a bit lost... :-(
Did you succed to use LoRaWan on your sensors ? What sketch did you use ? (found here or other)

Thanks,