CongducPham/LowCostLoRaGw #326: 32 bit image doesn't boot on Rpi4

tk5ep, 2023-07-13:
Hi,
I downloaded the Google drive 32 bit image and burned a SD card with Balenaetcher.
When inserted in a brand new Rpi4, nothing happens. No activity at all.

Looking at the manual installation page, it is an old Debian version with deprecated packages. As well as confusing informations.
There is even no guide on how to install the gateway software once downloaded on a working Debian distrib.

Can someone help ?
Thanks

CongducPham, 2023-07-13:
You should use the 64-bit image. regards,

tk5ep, 2023-07-14:
Merci,
Ca fonctionne. It works. :-)
What should I install as a **MINIMUM** on an existing Debian to make a simple LoRaWan gateway ?

Thank you,

tk5ep, 2023-07-14:
BTW, what are the login credentials for an SSH access for this 64 bit version ?
admin/loragateway doesn't work, nor pi/raspberry.

Thanks,

CongducPham, 2023-07-14:
it the same than on 32-bit: pi/loragateway