CongducPham/LowCostLoRaGw #327: Anyone wrote a BME sensor library ?

tk5ep, 2023-07-18:
Hi,

I plan to use a BME280 with the **Arduino_LoRa_LMIC_ABP_temp** example.
Did someone write a library to use with the BME280 sensor ? (or another sensor to get an example)

Thank you

dimtsam, 2025-01-11:
> Hi,
> 
> I plan to use a BME280 with the **Arduino_LoRa_LMIC_ABP_temp** example. Did someone write a library to use with the BME280 sensor ? (or another sensor to get an example)
> 
> Thank you

Hello
Did you find a solution? I am interested to use BME280 too.

tk5ep, 2025-01-13:
Hi,
I didn't continue my efforts with these softwares.
I never could make the single channel LoraWan gateway work.
The community is too small to get a valuable help. 
I'm written a small LoRa sensor that has a BME280 connected and is consuming 7 microAmps while asleep.

CongducPham, 2025-01-14:
I would advise you to check the new examples of the INTEL-IRRIS project which has better support for LoRaWAN. You can easily add the code for BME280 sensor:  https://github.com/CongducPham/PRIMA-Intel-IrriS

regards,

tk5ep, 2025-01-14:
Thanks,
I will have a look on that. Meanwhile, I'm using your pcb for my low power sensors. 

I see that you have a newer version.

Are there Gerber files for a pihat and the IRRIS gateway package ?

Regards,

dimtsam, 2025-01-14:
If someone find a solutiuon with the BME280 sensor please let me know, because I need help. I cant find the way to make it work by my own.
Regards

tk5ep, 2025-01-16:
@Dimtsam. What do you try to do ?
Does your gateway work ? LoRa, LoRaWan ?

I'm using some nodes with BME280, but NOT with the framework from CongducPham.
It's a Arduino pro mini with RFM95W and BME280 and LoRa, not LoRaWan.

dimtsam, 2025-01-16:
I didnt do anything more than following the guides. And it was my first time using Raspberry Pi. So I build some sensors (temp and humid) and upload the measurements to a Thingspeak Channel. The band is 868Mhz. 

I looked the example too, but I cant move further. The example uses an analog temp sensor. I cant figure out how to use the digital sensor that measures 3 different variables (Temperature Humidity Barometric), and send them to gateway. 
So I am looking for help

Regards
Dimitris

CongducPham, 2025-01-16:
You can look at the Arduino_LoRa_SX12XX_Generic_Simple_MultiSensors.ino example (with for instance the DHT sensor) to see how to handle a multi value physical sensor. The approach is to create a logical sensor for each of the measurement, e.g. Temperature, Humidity, Barometric.

Hope that helps.
regards,

dimtsam, 2025-01-17:
This is exactly the example I am working on. But I cant make any progress at all...
Regards