CongducPham/LowCostLoRaGw #328: Gateway : no rx found

tk5ep, 2023-08-01:
Hi,

I've build one of your low power device. It works and I have checked that it transmits OK.

Now I'm struggling with the gateway. I installed the latest 64 bit image on a Rpi4.
It loads and I can access to the admin page as well a SSH.

I connected a SX2718 module according to your readme files :
      RPI            Radio module
   GND pin 25----------GND   (ground in)
   3V3 pin 17----------3.3V  (3.3V in)
 GPIO4 pin  7----------RST   (Reset) 
CS/CE0 pin 24----------NSS   (CS chip select in)
   SCK pin 23----------SCK   (SPI clock in)
  MOSI pin 19----------MOSI  (SPI Data in)
  MISO pin 21----------MISO  (SPI Data out)

In radio.makefile I have : 
SX12XX=SX127X
CFLAGS+=-DPABOOST
CFLAGS+=-DMAX_DBM=14
CFLAGS+=-DBAND433
CFLAGS+=-DNSS=10 -DNRESET=7 -DRFBUSY=6

in raspberry.makefile : 
SPILIB=WIRINGPI
CFLAGS+=-DGETOPT_ISSUE

And run make_loragateway

Despite this, the Web admin page says : no low level status & no rx found

What can be wrong ?

I also don't have any data flow to connect to the declared gateway...

Thanks,

tk5ep, 2023-08-01:
Complement, when typing :
`sudo ./lora_gateway`

I get this error : 
```
Oops: Unable to determine board revision from /proc/cpuinfo
 -> Only Raspberry device is supported!
 ->  You'd best google the error to find out why.
```

tk5ep, 2023-08-01:
Bonjour,
Oui c'est bien cette image que j'ai téléchargée.

Yes, that's right. 
I have download it again this morning and reinstalled it, but it didn't help.
I also made an update from the admin page.

tk5ep, 2023-08-04:
No good idea ?

CongducPham, 2023-08-07:
No, I will try on my side to see where could be the issue.