CongducPham/LowCostLoRaGw #331: LoRa Gateway Not Receiving Messages When Bandwidth Set to 62 kHz

JacekDwojakPL, 2023-11-06:
Hello and thank you for creating this library! When configuring the LoRa gateway with a bandwidth of 62 kHz or below, I am experiencing issues where the gateway is not receiving messages as expected. This issue seems to be related to the bandwidth setting, as when I change it to other values, the gateway performs as expected. I am using Hope RF95W 868mhz radio module and trying to achieve maximum range out of the device, with different settings with bandwidth and spreading factor. Have you ever experienced such behavior?

CongducPham, 2023-11-07:
Hello, it is nomal, we did not implement the 62kHz band. It is possible that the low-level LoRa lib actually can do it but all our configuration scripts only take into account 125, 250 and 500kHZ band.

regards,

JacekDwojakPL, 2023-11-07:
which parts should I modify to have this band working? I am setting it here https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/SX127X_lora_gateway.h#L27 using constants from here: https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/SX12XX/SX127XLT_Definitions.h#L68-L77 (for exampleLORA_BW_041) on both gateway and transmitter. Code compiles ok and is running, I can see the output with correct setting. Transmitter transmits the message with longer time (which I assume is correct) but on the gateway side I can't recieve the data. Should I increase some timeout value? Thank you for your help

CongducPham, 2023-11-07:
If you are using the SX12XX lib (which support all SX12XX familly), you need to modify SX12XX_lora_gateway.cpp to accept and configure for the 62kHz case.

JacekDwojakPL, 2023-11-07:
sorry for not being clear - the 62kHz works fine, the problem starts when I try to set it lower. I checked the file for lora gateway you mentioned, and it seems to me that it takes command line argument to set the bandwidth and it works fine. Here is the output from simple lora transmitter configured to use bandwidth of 41kHz: 

`./simple_lora_transmitter 
MODE
<<< using wiringPi >>>
Lora Device found
Use PA_BOOST amplifier line
Set LoRa txpower dBm to 14

SX1276,865199936hz,SF12,BW41700,CR4:5,LDRO_On,SyncWord_0x12,IQNormal,Preamble_8
SX1276,SLEEP,Version_12,PacketMode_LoRa,Explicit,CRC_On,AGCauto_On,LNAgain_1,LNAboostHF_On,LNAboostLF_On

Reg    0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
0x00  99 81 1A 0B 00 52 D8 4C CC FC 09 10 3B 1C 00 00 
0x10  00 00 00 00 00 00 00 00 10 00 00 00 00 52 C4 64 
0x20  00 08 FF FF 00 00 0C 00 00 00 00 00 00 50 14 40 
0x30  00 03 05 27 1C 0A 03 0A 42 12 43 1D 00 AF 00 00 
0x40  30 00 12 24 2D 00 03 00 04 23 00 09 05 84 32 2B 

SX127X - Transmitter ready
Sending packet 0
Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,4946mS  PacketsSent,1
Sending packet 1
Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,4943mS  PacketsSent,2
Sending packet 2
Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,4942mS  PacketsSent,3
`

And here is the output of lora gateway program with command line argument:

`./lora_gateway --bw 41
<<< using wiringPi >>>
^$**********Power ON
^$LoRa BW 41000
^$Frequency 865200000
^$LoRa addr 1
^$Use PA_BOOST amplifier line
^$Set LoRa txpower dBm to 14
^$Preamble Length: 8
^$Sync word: 0x12
^$SX1276,865199936hz,SF12,BW41700,CR4:5,LDRO_On,SyncWord_0x12,IQNormal,Preamble_8
^$SX1276,SLEEP,Version_12,PacketMode_LoRa,Explicit,CRC_On,AGCauto_On,LNAgain_1,LNAboostHF_On,LNAboostLF_On
^$SX127X configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Will safely reset radio module every 100 packet received
^$Low-level gateway has downlink support
2023-11-07T16:20:09.077822
^$Low-level gw status ON
^$Low-level gw status ON
^$Low-level gw status ON
`

As you can see both of programs correctly reports usage of BW41, transmitter is sending the messages.
For comparison here are logs for both programs using BW62:

`./lora_gateway --bw 62
<<< using wiringPi >>>
^$**********Power ON
^$LoRa BW 62000
^$Frequency 865200000
^$LoRa addr 1
^$Use PA_BOOST amplifier line
^$Set LoRa txpower dBm to 14
^$Preamble Length: 8
^$Sync word: 0x12
^$SX1276,865199936hz,SF12,BW62500,CR4:5,LDRO_On,SyncWord_0x12,IQNormal,Preamble_8
^$SX1276,SLEEP,Version_12,PacketMode_LoRa,Explicit,CRC_On,AGCauto_On,LNAgain_1,LNAboostHF_On,LNAboostLF_On
^$SX127X configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Will safely reset radio module every 100 packet received
^$Low-level gateway has downlink support
2023-11-07T16:37:27.718217
^$Low-level gw status ON
--- rxlora[1]. dst=1 type=0x10 src=15 seq=0Timestamp has been written to timestamp.txt
 len=23 SNR=7 RSSIpkt=-62 BW=62 CR=4/5 SF=12
^p1,16,15,0,23,7,-62
^r62,5,12,865200
^t2023-11-07T16:38:00.071152*2549932016
��Hello World 1234567890*
^$-----------------------------------------------------
^$Check for downlink requests 2023-11-07T16:38:00.9711042550831968
^$NO NEW DOWNLINK ENTRY
--- rxlora[2]. dst=1 type=0x10 src=15 seq=1Timestamp has been written to timestamp.txt
 len=23 SNR=7 RSSIpkt=-58 BW=62 CR=4/5 SF=12
^p1,16,15,1,23,7,-58
^r62,5,12,865200
^t2023-11-07T16:38:04.367621*2554228485
��Hello World 1234567890*
`

`./simple_lora_transmitter 
MODE
<<< using wiringPi >>>
Lora Device found
Use PA_BOOST amplifier line
Set LoRa txpower dBm to 14

SX1276,865199936hz,SF12,BW62500,CR4:5,LDRO_On,SyncWord_0x12,IQNormal,Preamble_8
SX1276,SLEEP,Version_12,PacketMode_LoRa,Explicit,CRC_On,AGCauto_On,LNAgain_1,LNAboostHF_On,LNAboostLF_On

Reg    0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
0x00  99 81 1A 0B 00 52 D8 4C CC FC 09 10 3B 1C 00 00 
0x10  00 00 00 00 00 00 00 00 10 00 00 00 00 62 C4 64 
0x20  00 08 FF FF 00 00 0C 00 00 00 00 00 00 50 14 40 
0x30  00 03 05 27 1C 0A 03 0A 42 12 44 1D 00 AF 00 00 
0x40  30 00 12 24 2D 00 03 00 04 23 00 09 05 84 32 2B 

SX127X - Transmitter ready
Sending packet 0
Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,3299mS  PacketsSent,1
Sending packet 1
Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,3296mS  PacketsSent,2
Sending packet 2
Hello World 1234567890*  BytesSent,23  CRC,DAAB  TransmitTime,3296mS  PacketsSent,3
`

With these settings the transmitter and reciever works correctly. So my question is, is it the hardware issue that 62kHz is the boundary? Should I set frequency offset or another setting?

CongducPham, 2023-11-07:
OK, I see, I don't see obvious source of error here. There might be some parameters to set in order to use these very small BW value. As the time-on-air is very large (very small data rate) maybe some registers need to be correctly configured. Try digging into the SX127X doc (such as the one of RFM95W) and look at SX127XLT.cpp

Hope that helps.