CongducPham/LowCostLoRaGw #334: EBYTE E22 900M30S - no rx found

dolence, 2024-04-01:
I'm trying to make this gateway work with an EBYTE E22-900M30S module directly wired to the pins as described on the README without success.  I have tested this with the 32 and 64 bit images.

RPI PIN                              MODULE PIN
===============================
23 - GPIO11 (SPI_CLK)               SCK
21 - GPIO09 (SPI_MISO)            MISO
19 - GPIO10 (SPI_MOSI)            MOSI
24 - GPIO08 (SPI_CE0_N)          NSS
07 - GPIO04 (GPIO_GCLK)        NRST
3V3/GND 

I also have tried connecting BUSY and DIO1. What I'm doing wrong? Any thoughts?

[image]

CongducPham, 2024-04-02:
I think it is because the BYTE E22-900M30S is based on the SX1262 so you need to select the correct chip in radio.makefile

Also, the 1262 needs the BUSY pin.

regards,

dolence, 2024-04-03:
I already had the busy pin connect and the correct chip set:

```
#option are SX1272, SX126X, SX127X or SX128X
#SX12XX=SX1272
SX12XX=SX126X
#SX12XX=SX127X
#SX12XX=SX128X

#options are: -DPABOOST -DMAX_DBM=v -DBAND868|-DBAND900|-DBAND433 -DNRESET=6 -DNSS=10 -DRFBUSY=5
# default: NRESET=6 means Arduino pin 6 which will be translated into GPIO4 by ArduPi
# use NRESET=7 for WaziHat, translated into GPI17 by ArduPi
# default: NSS=10 means Arduino pin 10 which will be translated into GPIO8/SPI_CE0 by ArduPi
# default: RFBUSY=5 means Arduino pin 5 which will be translated into GPIO25/GEN06 by ArduPi

CFLAGS+=-DPABOOST
CFLAGS+=-DMAX_DBM=14
CFLAGS+=-DBAND915

#NRESET=0 is for wiringPi "GPIO. 0", correspond to GPIO17 (pin 11) (WaziHat)
#NRESET=7 is for wiringPi "GPIO. 7", correspond to GPIO4 (pin 7)
#RFBUSY=6 is for wiringPi "GPIO. 6", correspond to GPIO25 (pin 22) (SX1262/SX1282)
#NSS=10 remains the same for Arduino (arduPi) and wiringPi "CE0", correspond to GPIO8 (pin 24)
#
#LowCostLoRaGw SX12XX hat (C. Pham)
CFLAGS+=-DNSS=10 -DNRESET=7 -DRFBUSY=6
```
Did you had to make a jumper in between the module pins 7 and 8? I'm asking because as described in the datasheet, I had to make this to work on my Zephyr custom board without using TXEN and RXEN pins.

[image]
[image]

dolence, 2024-04-03:
Also, I have noticed the update script is not working. Wasn't supposed to be a trunk branch to make it work with SVN?
This link gives me a 404 error: (https://github.com/CongducPham/LowCostLoRaGw/trunk/gw_full_latest)

```
./update_gw.sh
updating gateway
removing any existing gw_full_latest folder (from previous update)
getting new gw_full_latest from github
using user-defined repository: https://github.com/CongducPham/LowCostLoRaGw/trunk/gw_full_latest
svn: E170013: Unable to connect to a repository at URL 'https://github.com/CongducPham/LowCostLoRaGw/trunk/gw_full_latest'
svn: E160013: '/CongducPham/LowCostLoRaGw/trunk/gw_full_latest' path not found
Failed to find a valid gw_full_latest folder for installation, exiting
```

CongducPham, 2024-04-03:
Thanks for your feedbacks. We actually didn't try with the EBYTE E22-900M30S but tested with NiceRF 1262 and there were no need for the jumper because they do not have such TXEN and RXEN pins.

For the github error, it seems that download paths have changed some months ago on github platform.,Will need to get the new path syntax & rules.

regards,

dolence, 2024-04-03:
SVN support was dropped. I had to manually upload the directory to my raspberry. Chirsptack isn't working too, even if I manually install as per instructions. What is the state of this project?

CongducPham, 2024-04-03:
If you take the SD card image, everything should work. However, yes, the project hasn't been updated for a while so the new Chirpstack distrib is probably not working.

dolence, 2024-04-03:
I took the SD card image, chirpstack isn't working. It is running but I can open it and no 8080 port is open.

CongducPham, 2024-04-06:
Will try on my side and get back to you. regards,

clow57, 2024-08-17:
Dolence did you manage to get this working with the  EBYTE E22-900M30S?

I am tryig to do the same.

dolence, 2024-08-19:
I just gave up and used a proper Pi based gateway.