CongducPham/LowCostLoRaGw #338: Endnode Arduino pro mini + inair9b unrecognized transceiver

donglectu, 2025-01-14:
I follow your Tutorial video https://www.youtube.com/watch?v=YsKbJeeav_M&t=1s but my endnode can't unrecognizable my module lora inair9b. Please help sir! 
[image]

CongducPham, 2025-01-14:
Check your wiring. With this old SX1272 lib the CS pin is pin 10. Check MOSI, MISO and CLK.

I would advise you to move to the more recent version Arduino_LoRa_SX12XX_Simple_temp where the lib is SX12XX which is better structured to define wiring of the lora module.

Hope that helps.

regards,

donglectu, 2025-01-14:
I try your suggest, use  Arduino_LoRa_SX12XX_Simple_temp source  and check my wiring, they all correct but it still not responding
[image]
[image]
[image]

CongducPham, 2025-01-14:
Looks like your MOSI and MISO are swapped on the pro mini.

donglectu, 2025-01-14:
I swap them, I also change another inair9b but nothing happened.

CongducPham, 2025-01-14:
Not sure to understand how you wire the VCC and the GND of the radio module. Maybe it is better to connect these pin to the 3.3v and GND of the pro mini

CongducPham, 2025-01-20:
Yes, but you need to enable LoRaWAN option for the gateway. By default, it is raw LoRa with our own packet header.

If you need to use LoRaWAN, you can look at the INTEL-IRRIS project, the updated sensor code and the WaziGate gateway: https://intel-irris.eu/results

regards,

donglectu, 2025-01-24:
when i try Arduino_LoRa_SX12XX_temp code. Serial cant read 

[image]

donglectu, 2025-01-24:
do i need to uncomment #define STRING_LIB

CongducPham, 2025-01-24:
No, no need to uncomment #define STRING_LIB. The behavior of your node is strange. I don't see any obvious reason.

donglectu, 2025-01-24:
this is my configuration 
#define WITH_EEPROM
////////////////////////////
//add 4-byte AppKey filtering - only for non-LoRaWAN mode
//#define WITH_APPKEY
////////////////////////////
//request an ack from gateway - only for non-LoRaWAN mode
//#define WITH_ACK
////////////////////////////
//if you are low on program memory, comment STRING_LIB to save about 2K
//#define STRING_LIB
////////////////////////////
#define LOW_POWER
#define LOW_POWER_HIBERNATE
#define SHOW_LOW_POWER_CYCLE
////////////////////////////
//Use LoRaWAN AES-like encryption
#define WITH_AES
////////////////////////////
//Use our Lightweight Stream Cipher (LSC) algorithm
//#define WITH_LSC
////////////////////////////
//If you want to upload to LoRaWAN cloud without pure LoRaWAN format you have to provide a 4 bytes DevAddr and uncomment #define EXTDEVADDR
#define EXTDEVADDR 
////////////////////////////
//this will enable a receive window after every transmission, uncomment it to also have LoRaWAN downlink
#define WITH_RCVW
////////////////////////////
//normal behavior is to invert IQ for RX, the normal behavior at gateway is also to invert its IQ setting, only valid with WITH_RCVW
#define INVERTIQ_ON_RX
////////////////////////////
//uncomment to use a customized frequency. TTN plan includes 868.1/868.3/868.5/867.1/867.3/867.5/867.7/867.9 for LoRa
#define MY_FREQUENCY 868100000
//#define MY_FREQUENCY 433170000
////////////////////////////
//when sending to a LoRaWAN gateway (e.g. running util_pkt_logger) but with no native LoRaWAN format, just to set the correct sync word
#define PUBLIC_SYNCWORD 0x34
////////////////////////////

///////////////////////////////////////////////////////////////////
// ADD HERE OTHER PLATFORMS THAT DO NOT SUPPORT EEPROM
#if defined ARDUINO_SAM_DUE || defined _VARIANT_ARDUINO_DUE_X_ || defined __SAMD21G18A__
#undef WITH_EEPROM
#endif

///////////////////////////////////////////////////////////////////
// ADD HERE OTHER PLATFORMS THAT DO NOT SUPPORT LOW POWER LIB
#if defined ARDUINO_SAM_DUE || defined _VARIANT_ARDUINO_DUE_X_
#undef LOW_POWER
#endif
///////////////////////////////////////////////////////////////////

///////////////////////////////////////////////////////////////////
// CHANGE HERE THE NODE ADDRESS 
uint8_t node_addr=8;
//////////////////////////////////////////////////////////////////

///////////////////////////////////////////////////////////////////
// CHANGE HERE THE TIME IN MINUTES BETWEEN 2 READING & TRANSMISSION
unsigned int idlePeriodInMin = 3;
unsigned int idlePeriodInSec = 0;
///////////////////////////////////////////////////////////////////

#ifdef WITH_APPKEY
///////////////////////////////////////////////////////////////////
// CHANGE HERE THE APPKEY, BUT IF GW CHECKS FOR APPKEY, MUST BE
// IN THE APPKEY LIST MAINTAINED BY GW.
uint8_t my_appKey[4]={5, 6, 7, 8};
///////////////////////////////////////////////////////////////////
#endif

///////////////////////////////////////////////////////////////////
// ENCRYPTION CONFIGURATION AND KEYS FOR LORAWAN
#ifdef LORAWAN
#ifndef WITH_AES
#define WITH_AES
#endif
#endif
#ifdef WITH_AES
#include "local_lorawan.h"
#endif

///////////////////////////////////////////////////////////////////
// LORAWAN OR EXTENDED DEVICE ADDRESS FOR LORAWAN CLOUD
#if defined LORAWAN || defined EXTDEVADDR
///////////////////////////////////////////////////////////////////
//ENTER HERE your Device Address from the TTN device info (same order, i.e. msb). Example for 0x12345678
unsigned char DevAddr[4] = { 0x27, 0xFC, 0x34, 0x16 };
///////////////////////////////////////////////////////////////////

//Danang
//unsigned char DevAddr[4] = { 0x26, 0x04, 0x1F, 0x24 };

//Pau
//unsigned char DevAddr[4] = { 0x26, 0x01, 0x17, 0x21 };
#else
///////////////////////////////////////////////////////////////////
// DO NOT CHANGE HERE
unsigned char DevAddr[4] = { 0x00, 0x00, 0x00, node_addr };
///////////////////////////////////////////////////////////////////
#endif

///////////////////////////////////////////////////////////////////
// ENCRYPTION CONFIGURATION AND KEYS FOR LSC ENCRYPTION METHOD
#ifdef WITH_LSC
#include "local_lsc.h"
#endif

///////////////////////////////////////////////////////////////////

///////////////////////////////////////////////////////////////////
// IF YOU SEND A LONG STRING, INCREASE THE SIZE OF MESSAGE
uint8_t message[80];

[image]

///////////////////////////////////////////////////////////////////

[image]

[image]

[image]

CongducPham, 2025-01-24:
As I said, I don't see any obvious reason for your serial monitor output to look like that. The baud rate looks fine at 38400.

donglectu, 2025-01-24:
Hello sir, I have successfully uplinked to ttn, I have selected the processor for version 3.3v

[image]

donglectu, 2025-01-24:
how long for another uplink ,  should i  set aes to false or true to get better results

[image]

donglectu, 2025-01-24:
How to make my gateway appear on ttn map? https://www.thethingsnetwork.org/map 

[image]

donglectu, 2025-02-03:
mkr wan 1300 do have cs pin so how to set up it with SX127X_RadioSettings.h ?