CongducPham/LowCostLoRaGw #339: Can Waziup GW be configure for a private chirpstack server & 433 MHz ?

tk5ep, 2025-02-01:
Hi, I've installed the latest Waziup GW v 2.3.2 with your RFM95W board.

Can it be configured for the 433 MHz band and a private Chirpstacl/MQTT server ?

Thanks,

CongducPham, 2025-02-01:
Have you installed the setting from the PRIMA INTEL-IRRIS project?

tk5ep, 2025-02-01:
Hi, I installed the image found on the Waziup site.
I'm a bit confused as going on one site brings you back and forth on your Github, some links are broken and finding the right documentation is a bit difficult.  :(
What image should I use ?

Thanks,

CongducPham, 2025-02-01:
LowCostLoRaGw is now kind of obsolete. It still can be used if you want a real low-level single-channel gateway where you can control everything. We are still using it for instance for some LoRa 2.4GHz prototyping.

For more operation deployment, we developed in the PRIMA INTEL-IRRIS project a new gateway based on the WaziGate. BUT, we have our own SD card image, so, do not get the generic WaziGate SD card image because there are a lot of default configuration in our own SD card image to make sure everything works out-of-the-box for the INTEL-IRRIS ecosystem.

Tou have everything here:

- https://intel-irris.eu/results
- https://intel-irris.eu/tutorials-slides

Please for questions related to the INTEL-IRRIS setting, please use the INTEL-IRRIS GitHub https://github.com/CongducPham/PRIMA-Intel-IrriS to ask questions

regards,

tk5ep, 2025-02-01:
Thanks, 
I have downloaded and burned the image.
I changed the band to eu433 and could log into the web page.

Now trying to find out how to set a private chirpstack server and change the LoRa settings (SF, rate, etc...).

Will post on the Github you mentioned.
Thanks for the help.