CongducPham/LowCostLoRaGw #340: The problem of gateway cannot send the packet to TTN

ho-hoang-phu, 2025-03-11:
I created a node to send data from Arduino Pro Mini, follow your instructions, but replace it with an angular accelerator sensor MPU6050, GateWays from Raspberri Pi3 and Inair9B module, I am having a problem with Node to send data to Gateway received but cannot be sent to TTN, below is the Setting Node section, please help me, thank

[image]

[image]

[image]

[image]

[image]

[image]

ho-hoang-phu, 2025-03-15:
> I created a node to send data from Arduino Pro Mini, follow your instructions, but replace it with an angular accelerator sensor MPU6050, GateWays from Raspberri Pi3 and Inair9B module, I am having [...]

closed 2025-03-15 by ho-hoang-phu

CongducPham, 2025-03-16:
We are not maintaining https://github.com/CongducPham/LowCostLoRaGw/ anymore. I would advice you to use the gateway version we developed in INTEL-IRRIS: https://github.com/CongducPham/PRIMA-Intel-IrriS
regards,